prusa3d/Prusa-Firmware #2761: [BUG] After power off Z-Axis Increases upon loading and unloading unexpectedly

sgano, 2020-06-25:
MK3S
3.9.0
**SD card or USB/Octoprint**
 USB/Octoprint

**Describe the bug**
When the printer is first turned on, when unloading or loading filament the z-axis is raised.  This causes it to hit the top of the printer after tall prints. 

In the standard g-code the extruder is already raised up well off the bed when a print is completed; so if a printer is turned off after a print when it is turned back on -- the extruder does not need to be automatically raised when loading and unloading filament before it is homed.    This has caused me a lot of headache in hitting the top of the printer. 

  
**To Reproduce**
  After a tall print, turn the printer off.  Turn it back on then preheat the bed and nozzle - and then unload and load filament -- the printer will have raised the z-axis twice -- potentially hitting the top of the printer. 
  
**Expected behavior**
   I can raise the z-axis if needed -- I would not expect the z-axis to raise when loading or unloading filament automatically. 
  
**G-code**
NA

**Video**
NA

leptun, 2020-06-26:
In firmware 3.9.0, if the Z axis hits the top holders when raising, it will stop instantly and prevent further movement upwards. The printer looses all track of positioning once the firmware is reset. That’s why homing exists.

i0n0s, 2020-06-30:
I can confirm the described behaviour. 
The raise on the unload is a regression in the 3.9 firmware. Previous versions didn't raise on unload.
The raise on the load can be discussed. It is comfortable if the nozzle was previously too low, so that you can easily remove the extruded filament. But from what I recall, this should be the case in nearly every case. Only if the user manually lowered the z-axis, then it will be needed. But this can only happen if the axis is calibrated, otherwise it can't get lowered than the starting position.
If I forgot about some scenario, please write about them.

wavexx, 2020-06-30:
Yes, this is a new feature in FW 3.9, which is also done while preheating to avoid leaving marks on the PEI sheet.
Raising is done if Z is not homed yet, or if we know Z is low enough.

However, stallguard is activated while raising, so the "hit" shouldn't ram the axis.

i0n0s, 2020-06-30:
Well, I've just tried it and it doesn't work like you describe it:
It will first preheat, then when at temperature, after pressing the button, it will raise and unload. So there will be marks on the sheet since the nozzle was at temperature before raising.
The same behaviour occurs when loading new filament on an empty, turned off, printer:
It will ask for the temperature, waits till the temperature is reached, waits for user confirmation, starts extruding and then raises the z-axis.
For completion, I also tried to preheat manually by selecting the matching profile from the menu. In that case it raised the z-axis and then started to heat.

You should move the raise before the heat in the other cases. And for convenience: Also set a flag that it was raised, so the following steps won't raise again.
When changing the filament via the actions "Preheat", "Unload" and insertion of the filament, the z-axis will be raised 3 times.

wavexx, 2020-06-30:
Can you record a video?

i0n0s, 2020-06-30:
[20200630.zip](https://github.com/prusa3d/Prusa-Firmware/files/4853807/20200630.zip)

Sadly you can't read the screen on the video.
I've selected "Unload filament" -> "PET" then waited till the temperature reached 220°C. It was that fast because the hotend was still hot. But the temperature of the hotend doesn't make a difference.
The extruder firsts starts moving after I've clicked the confirmation button for the unload.

sgano, 2020-07-01:
I agree if the intent is to prevent possible damage to the PEI sheet/bed then I would expect that when loading filament it would move up first before loading any amount of material.  Currently it loads some, moves up, then loads more. 

Personally don’t like this behavior as it’s unexpected, when selecting loading or unloading filament from the menu a user doesn't expect z-axis movements to occur in addition to loading and unloading.  This seems to protect the bed in only a very small number of use-cases (e.g., power off when the nozzle is still on the first layer).  

If I may suggest a potential compromise; don’t move up 20mm when loading and unloading filament (prior to the machine being homed) but instead move only 5 mm as that is plenty to prevent damage - and would result in less of an unexpected movement.

i0n0s, 2020-07-01:
The movement of 20mm is a good one for loading. I need to remove the extruder filament after the load and my finger is slightly larger than 5mm and I prefer to have a little distance to the hot nozzle too. So I can understand the lift to 20mm (or about 20mm if unknown position) when loading.
For heating and unloading the lift distance can be 5mm or less.

wavexx, 2020-09-22:
Took me a while, but I've got a look at this. It's a bit odd, but it does behave differently depending if the printer has been reset or not. When the printer has been reset, but not homed yet, we don't know where the extruder is, so we attempt to raise the nozzle anyway.

In this condition, while moving up for loading, the move *has* to be split, because we turn on crash-detection on the Z axis to ensure we stop quickly if we hit the top.

Raising is done when preheating to avoid leaving heat "scars" on the steel sheet.

But when loading, this change was introduced because we plan to extrude a little bit to just prior to unload to ensure the unload doesn't jam:

https://github.com/prusa3d/Prusa-Firmware/issues/2761

i0n0s, 2020-10-03:
I've just tested this issue with the new 3.9.1:
The behaviour didn't changed at all. And in the "Unload procedure" I still get the raise _after_ the temperature has reached the target temperature. This would leave a heat scar if done at Z=0. All done on a freshly reset printer which doesn't know its Z position.

sgano, 2020-10-03:
There is another case to consider here as well -- if there is a power outage and the nozzle is then "frozen" to the part; I want the ability to heat the nozzle without it raising up while cold and potentially damaging something.  Then when it is up to temperature I will raise it up to properly unstick it from the part.

wavexx, 2020-10-03:
I'll try this again and try to make a video as well.
@leptun Thinking out loud, could it be that some Z motors need some back-off before stallguard properly detects the stall, similarly to X? Trouble is, we can't back off Z until it has homed.

Just as a background information, the stall-detection was added because there were also problem when printing (not just loading/unloading filament) just after a restart if Z was already at the top. The homing procedure used to move Z up twice without checking, ramming the Z tops in the same way.

Could you check if that works? Move Z to the upper limit manually, then reset the printer, and try to "Auto home". Does it also ram the axis or does it properly stop as soon as the tops are hit?

wavexx, 2020-10-28:
@i0n0s Turns out I was doing "preheat" out of habit without even thinking. Looking at the video, going through load/unload, the raise is indeed done *after* preheating. I created PR #2875 which should fix this. Can you give it a test?

@sgano good point. In FW 3.9 the extruder will try move away from the print, as originally intended. AFAIK, the only case where the PSU hasn't enough juice to perform the move is when the MMU is also attached :(, something which still needs to be fixed.

However, should the nozzle be stuck for whatever reason, setting the nozzle temperature directly doesn't perform any move.

i0n0s, 2020-10-28:
Ok, gave it a try and tested the following scenarios on a reset printer:
*Preheating was raising by 10mm before heating.
*Loading the filament will raise the extruder by 50mm before heating.
*Unloading the filament will raise the extruder by 20mm before heating and loading the filament directly after will raise another 30mm.
So the raise is done before the preheating which works correct. But the different heights and the additional raise doesn't make sense for me.

wavexx, 2020-10-28:
I originally did this to make it as fast as possible. 10mm was enough to avoid marks on PEI even with the nozzle at 290C.
Unloading combines a final move so that the final height is the same (50mm). Fully agree that it should be the same.

But I need check (or if you could also check) if unloading above 50mm doesn't raise _again_, or if we need to change that too. That was some hard-coded behavior that I simply left in place.

i0n0s, 2020-10-28:
Calibrated printer: Z=40mm:
*Unload: No raise occurred.
*Loading: Raise by 10mm during extrusion.

Uncalibrated printer: Z=51.15? (manually moved Z by 51mm):
*Unload: No raise occured.
*Loading: Raise by 10mm during extrusion.

Uncalibrated printer: Z=0.15?
*Unload: Raising by 20mm
*Loading: Raising by 30mm
*Unload: No raise occured.

So the behaviour above 50mm is as you expected. 

And I discovered a bug in the loading routine which is sadly not reproducible: It will start extruding after confirming that the filament was extruded correctly. But this is another issue and needs to be reproducible first.

wavexx, 2020-10-29:
I pushed one more change to make them both 50mm. This way unloading+loading doesn't move it twice.

PR #2875 (2021-01-11): Raise Z while preheating when auto/[un]loading the filament
Preheating already raised Z to avoid scorching the PEI sheet, as does
filament loading/unloading to allow for excess material to be removed.

However, when loading/autoloading/unloading via the LCD with a cold
nozzle the preheating menu is performed before the carriage is raised,
leaving the carriage close to the sheet while heating the nozzle.

Pre-raise the carriage already while waiting, so that the subsequent
move is automagically skipped.

Set bFilamentWaitingFlag only once to perform both the LCD
initialization and raising to th [... cut]
commit 5f23474c (2021-01-11): Raise Z while preheating when auto/[un]loading the filament

Preheating already raised Z to avoid scorching the PEI sheet, as does
filament loading/unloading to allow for excess material to be removed.

However, when loading/autoloading/unloading via the LCD with a cold
nozzle the preheating menu is performed before the carriage is raised,
leaving the carriage close to the sheet while heating the nozzle.

Pre-raise the carriage already while waiting, so that the subsequent
move is automagically skipped.

Set bFilamentWaitingFlag only once to perform both the LCD
initialization and raising to t [... cut]
closed 2021-01-11 by DRracer