prusa3d/Prusa-Firmware #2762: [BUG]Too long bed temperature stabilization

espr14, 2020-06-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - N/A
**MMU upgrade firmware version N/A

**SD card or USB/Octoprint**
  SD

**Describe the bug**
  Printer waits very long for bed temperature stabilization (M190).
  
**To Reproduce**
1. Preheat printer for PET
2. Run M190 S85
3. It took me around 30 seconds to proceed.
  
**Expected behavior**
It should take few seconds.
  
**G-code**
M190 S85

**Video**
N/A

In original Marlin this is caused by `+=` that doubles the waiting time. The line should be omitted or (half of) the time should be subtracted.
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/temperature.cpp#L3320

wavexx, 2020-09-22:
I noticed this takes longer than it should be sometimes. I guess the same happens for the extruder? (the major difference is that the extruder has less lag, thus is just faster overall)

espr14, 2020-09-22:
I was searching for Marlin's algorithm for the bed but it's not in the FW. Extruder stabilization time is 5 seconds (or 10 at most) which is OK. You can close the issue. Or let me know once you'll be using Marlin's algorithm and constants and I'll fix it.

closed 2023-10-23 by espr14