prusa3d/Prusa-Firmware #2763: [BUG] FINDA sensor goes N/A during retraction

scooper4711, 2020-06-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches. - Yes

**Printer type** - MK3S
**Printer firmware version**-  3.9.0-3421

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6-372

**SD card or USB/Octoprint**
  SD

**Describe the bug**
  During retraction, the FINDA sensor will consistently report N/A on the screen. Backing down to the previous release of firmware eliminates this problem.
  
**To Reproduce**
  Load a MMU print with filament changes.
 While printing, switch to display the sensor information.
 Observe the FINDA sensor during retractions - as soon as retraction starts, it switches to N/A and stays there. It returns to a '1' state only upon successful load of the next filament. It never goes to 0.
  
**Expected behavior**
  The FINDA should remain at 1 until the filament is no longer in the switcher (or whatever its called). It should remain 0 until the next filament is detected.
  
**G-code**
  n/a

**Video**
  Video is too large. I'll try to edit it down.

.

scooper4711, 2020-06-28:
[IMG_3140.mov.zip](https://github.com/prusa3d/Prusa-Firmware/files/4842491/IMG_3140.mov.zip)
Movie attached.

scooper4711, 2020-06-28:
[IMG_3138.mov.zip](https://github.com/prusa3d/Prusa-Firmware/files/4842494/IMG_3138.mov.zip)
Second movie showing that the N/A stays until the filament is loaded.

Ondrejicek, 2020-07-29:
I can confirm, it happens always on my unit.

shadowa2j, 2020-08-01:
I am having the same issue, been pulling my hair out for the last couple hours trying to troubleshoot. Guess I will be downgrading my firmware for the MMU tomorrow and trying that. I did have some successful prints after upgrading the firmware, but now I can't even get one started.

Botond24, 2020-08-01:
Having the same issue, thought it was my printer hardware, but no.
Thank you

Sebas44, 2020-08-14:
Having the same issue. What a pity I find this only after I changed the FINDA probe to no effect.

Forfelet, 2020-09-25:
Hello everybody,

I used currently the last firmware 3.9.1 and I have the same issue. When filament is unloaded, N/A is displayed as FINDA value and remain this value during change filament. 
When the next loading is done properly, I have no problem with that. Value is correctly updated to 1. 
But, if next loading is not ok, printer cannot be refind a great configuration (all led are blinking).
With many reset, mmu is totally lost (all led are blinking and the selector position is not correct according to gcode). 
After multiple reset and my help, finally the selector find the good position and filament is loaded. But, the printer launch a filament runout procedure. It goes to the right corner to unload and reload new filament before to return on the purge tower.

So, I think that to have a value N/A during the filament change is not a good thing.
After a mmu reset, printer and mmu's configuration don't match.

I think downgrade my printer to the FW3.8.1.

AndreasBrauckmann, 2020-10-11:
I had this problem too, i don't understand why this problem occurred when changing firmware, coincidence?

- MMU reset (no solution)
- FINDA change (no solution)
- Firmware downgrade (sometimes it worked .. sometimes not)
- **tightened the two screws (it was that easy for me)**

**These two screws (see red arrows) must be firmly seated on the shaft.**

[image]

peon374, 2020-11-06:
Mine does the same

glowfishDE, 2020-11-06:
i also have the same issue and the MMU2S is unable to unload filament from time to time

katony411, 2021-03-11:
Same. I thought it was deliberate when troubleshooting. that you turned the finda probe off when retracting. This is causing fail print because the filament fails to fully unload. and then stays on NA until reboot/ Something thing random happens. Now I know it has to be 1 or 0.... rechecked wires. all seems plugged in tight(so should not be shaking). and get 0/1 when rebooting.

bryn51, 2022-07-26:
It’s now July 2022 and yet I see this same bug on my MMU2. In the middle of a print during a fikament change the finda sensor goes to n/a and the printer goes cactus(throws up its hands confused). I try to intervene by manual load of filament but even if I get one going it fails on the next.
This seems to be related to filament position 1 where t strain no the finda cable is greatest.
So I have rashly assumed this is a wiring issue and ordered a replacement finda.
but there are no comments on here from Prusa Research after over 2 years of people reporting this issue. Looks like they just do not care. Prefer to concern with new products than maintaining the old.

Prusa-Support, 2022-08-02:
Thank you all for your feedback, and for your patience especially.

I'm sorry to hear that the imprecise FINDA status was misleading for some of you.

Please run troubleshooting on the FINDA and the sensors themselves while the printer is not operating. The status of the sensors on the screen 'sensor info' may not be refreshed in real-time during the printer's operation, especially the FINDA's one, which is processed through a secondary board.
For help and troubleshooting, our Customer Support remains available.
https://help.prusa3d.com/article/customer-support_2287

Please notice, this doesn't mean the sensor reading is unreliable: the output of the sensor during the combined print operations are read and processed before even being displayed in 'sensor info'. In other words, the operations don't realy on 'sensor info', which can still be useful for troubleshooting if the printer is not "busy".

Some of you may have noticed that the "N/A" status didn't occur in the previous firmware release while printing. That is because the status sometimes was not updated during the print, even if the FINDA status was not available.

This is apparently not a bug, but an enhancement request.
The behavior of the FINDA in 'sensor info' should be improved in the next firmware. It may take some time.

Michele Moramarco
Prusa Research

bryn51, 2022-08-02:
Thanks Michelle for the info

However my own observation and experience lead me to a different conclusion.

What I found was that printing if the “test flower” print failed 20-25% after commencement always on change of the 5th filament and always showing n/a on the finda probe.

I looked at several other things before finally changing the finda probe.
And after that no longer does the test flower print fail and no longer showing n/a status.

You seem to be saying that finda displaying n/a is not an indication of a failed finda probe, but (since the above  symptoms occurred several times) it seems otherwise to me.

Cheers
Bryn Parrott

bryn51, 2023-10-23:
I think it is stale because over-ridden by MMU3 development. Lets see if it raises its head again.

closed 2024-01-01 by github-actions[bot]