prusa3d/Prusa-Firmware #2764: [BUG]fan check occurs after pausing print with fan checks off in settings

Lemlurkerman, 2020-06-29:
**Printer type** - MK3S,
**Printer firmware version**-  one version behind, will double check

**SD card or USB/Octoprint**
  Sd

**Describe the bug**
  After pausing the print the printer checks the hot end fan. My fan has a failed sensor cable (but works) this causes the print to error out after pausing and refuse to continue
  
**To Reproduce**
  Have sensor less /sensor failed hot end fan. Disable fan checks, start printing and then pause, upon restarting print will fail
  
**Expected behavior**
  Fans will not be checked and print continues
  
**G-code**
  Occurs on all gcode

StevenInc, 2020-10-06:
I recently experienced a very similar error.
We updated all our printers to 3.9.1 and we had 8 printers on a rack running.
We put all 8 on pause for about 20 minutes while we moved the rack.
And when we went to restart the printers 4 of the 8 printers had a Fan Check Error.
see photo: https://photos.app.goo.gl/W2SCr8MkZFZrvPKe6

gudnimg, 2022-03-13:
Is this still an issue for you? I don't see how this error could appear with the "Fan Check" setting off. If this is an issue still, then it must be somehow `fans_check_enabled` is not set correctly. 🤔

closed 2023-10-31 by github-actions[bot]