prusa3d/Prusa-Firmware #2766: [BUG]Weird behavior after stopped print

espr14, 2020-06-29:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0 - 3421

**MMU Upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  After a print is stopped during printing, it goes up only. After that, a move in axis X causes moves in XYZ axes. Happened twice with different gcodes.
  
**To Reproduce**
   You can try the attached gcode. I§m not sure if it will happen.
1. Run the gcode.
2. Pause and resume few times.
3. Stop the print.
4. If the extruder goes up only, it§s the bug.
5. Go to Settings -> Move axis -> X and shift it by few milimeters.
6. Extruder goes down and makes more moves.
  
**Expected behavior**
  After a print, the extruder should go up and the bed should go to the front
  
**G-code**
[weird.zip](https://github.com/prusa3d/Prusa-Firmware/files/4848161/weird.zip)

**Video**
  Maybe next time.

wavexx, 2020-06-30:
Did you notice if it does that also if you don't pause inbetween?

espr14, 2020-06-30:
I think the first time it happened (different gcode), it was not paused.

espr14, 2020-06-30:
Video after print is stopped:
https://drive.google.com/file/d/1CylSpBPsiOg7KxRofCWtwg-SOIxFM2Ob/view?usp=sharing

wavexx, 2020-06-30:
> https://drive.google.com/file/d/1CylSpBPsiOg7KxRofCWtwg-SOIxFM2Ob/view?usp=sharing

This doesn't seem to be public

espr14, 2020-06-30:
Now should be.

wavexx, 2020-06-30:
Oh yes, I've seen this behavior before. Was the print ever paused in this particular case? (just asking to make sure, since pause and stop do share some code).

Do you think that this model makes it happen more than the others?

I never thought of using "move axis" after doing stop. It seems that this helps in returning control to a previous state of the planner before stop was called, so in all likelihood this is another case where 'stop' is being nested recursively.

Including also @leptun

espr14, 2020-06-30:
This case was paused twice (I think). Since I haven't seen it before, it could be caused by the specific G code. I use SuperSlicer now instead of PrusaSlicer (used before). There could be something that could trigger this behavior.

wavexx, 2020-06-30:
Ok - that's important. There means that pause could also be the culprit.

wavexx, 2020-09-22:
@espr14 we fixed a bug in 3.9.1 (https://github.com/prusa3d/Prusa-Firmware/issues/2766) which I suspect could *also* be the cause of this (in fact, I suspect that issue could be behind many planning oddities we had, due to it's rarity!).

Could you give the latest FW a test and keep an eye on this? It's really hard to reproduce, so when you're using stop or pause, if you can keep a record of what happened during the print (if it was started after being stopped, for example) it would help a lot.

espr14, 2020-09-22:
I'm selling MK3S so I won't be able to check this issue with 3.9.1. You can close the issue.

JohnnyDeer, 2020-12-01:
> You can close the issue.

OK :+1:

closed 2020-12-01 by JohnnyDeer