prusa3d/Prusa-Firmware #2767: [FEATURE REQUEST]Cold Pull Routine in the Printer Menu

livingstonjjks, 2020-06-29:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - All
**Describe the solution you'd like**
  A guided Cold Pull routine in the printer menu please. This will speed up and help new users clean the nozzle.

5jvm0u4, 2021-03-20:
Just about to suggest the exact same thing, this gcode is my way to add this feature to my printer by leaving this in my SD card.

[ColdPull_Script.txt](https://github.com/prusa3d/Prusa-Firmware/files/6175779/ColdPull_Script.txt)

ReeceWeb, 2021-09-20:
Was about to request this feature too…
At the very least, add two new entries to the Preheat menu:
(1) ColdPull High (270°C Nozzle / 0°C Bed)
(2) ColdPull Low (85°C Nozzle / 0°C Bed)

gotdibbs, 2021-11-10:
This would be awesome!

@5jvm0u4 your solution looks great but it looks like it sets the temperatures to something other than what I expected for the mk3s based on their docs (270C / 85C), is your gcode for a different printer or is there another reason?

5jvm0u4, 2021-11-11:
> This would be awesome!
> 
> @5jvm0u4 your solution looks great but it looks like it sets the temperatures to something other than what I expected for the mk3s based on their docs (270C / 85C), is your gcode for a different printer or is there another reason?

I wasn't paying attention to the docs😂 It's purely my experience with a specific PLA brand in Taiwan which I got very consistent result out of, and while reheating, I'll start pulling as soon as the temp passes 70C, setting it to 215C is just for convenience for the upcoming print.
I understand the reason why 270C is needed, but that means cooling needs more waiting time and I'm inpatient, and also so far 215C hasn't cause me any problem.
It's tuned for my stock Prusa i3 MK3s, yes.
[image]

gotdibbs, 2021-11-11:
Awesome! Thanks for the info @5jvm0u4

scooper4711, 2023-10-23:
I would still like to see this. Commenting to remove the stale label.

3d-gussner, 2023-10-25:
@scooper4711 
Different materials need different temperatures for a cold pull.
Boiling PLA with 285° can cause more issue than the cold tries to solve, and adding a new menu for all the different materials uses lot of precious resources.
We are very limited on resources and this barely used feature will not be implemented as gcodes like above are a good way to do that.
 
Sorry.
Closing issue.

closed 2023-10-25 by 3d-gussner