prusa3d/Prusa-Firmware #2768: [BUG] Left side on a full-bed print can show artifacts

wavexx, 2020-06-30:
**Printer type** - MK3/MK3S
**Printer firmware version**-  3.8.1+ (verified on 3.9.0)
**MMU Upgrade** - N/A (MMU2S)

**Describe the bug**
In some circumstances, doing a print which spans the entire X axis length might introduce artifacts on the leftmost side of the print, as shown here:

[image]

Depending on the height and Y position some patterns can become apparent:

[image]

The same model shifted to the right makes the problem disappear.

There is likely a problem with XYZ calibration in this setup, where the combination of PINDA offset and bed origin is causing an underflow on the X coordinate at a low level (probably after the world2machine correction) already at 0.25mm on the X axis.

The extrusion direction in the outer perimeter shown is going towards the front of the print, where the problem gets progressively worse (either due to skew/rotation correction or stepping along the underlying unsigned counters).

**To Reproduce**
Extrude along the left edge in some printers, hinting that our internal machine origin is too close at least to the left side.

Details of XYZ cal to be added later @vertigo235 
  
**G-code**
This is a reduced version of the full print which is left-aligned to the bed and still shows the artifact on the printer it was reported to: MK3S FW 3.9.0 + MMU2S

The actual model is likely irrelevant. A box aligned to the left should produce the same results.

[Utensil_Tray v4.3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/4850826/Utensil_Tray.v4.3mf.zip)

wavexx, 2020-06-30:
Shown in the XYZ cal:

```
Y distance from min:
Left:  11.88mm
Right: 11.96mm

Measured skew: 0.04

[0;0] point offset:
X -0.96mm
Y  1.38mm
```

MattBlissett, 2021-10-17:
I think this is probably the same problem I noticed, though I have an easier example.  Firmware 3.10.0.

Create a very wide cuboid (mine is 243.48 × 203.48 × 0.4mm).  Set skirt loops to three, set 1 perimeter, 0 top/bottom layers, 0 infill, I'm only interested in the skirt.

The skirt loops overlapped on the left side, though the right side is normal:

[image]

For my actual print, this led to clicking/skipping of the extruder, as it tried to print over the existing skirt lines.  A 250mm object would end up as 249mm, with overprinting (assuming > 1 perimeter) on the left side, as wavexx has.

Now change the bed shape, to set the X origin to -1 (recentre the cuboid).  This prints correctly, although rather close to the right edge of the steel sheet:

[image] 
[image]

In the first case, the G-code for the skirt: 

```
;TYPE:Skirt/Brim
;WIDTH:0.42
...
G1 X3.26 Y209.871 E7.63414
G1 X1.74 Y209.478 E.04923
G1 X.602 Y208.396 E.04923
G1 X.129 Y206.74 E.054
G1 X.129 Y3.26 E6.37997
G1 X.522 Y1.74 E.04923
...
```

And in the second case:

```
;TYPE:Skirt/Brim
;WIDTH:0.42
G1 X4.26 Y210.871 E7.63414
G1 X2.74 Y210.478 E.04923
G1 X1.602 Y209.396 E.04923
G1 X1.129 Y207.74 E.054
G1 X1.129 Y4.26 E6.37997
G1 X1.522 Y2.74 E.04923
```

I think the commands to move to X<1 are ignored, or rounded to 1 (or some other close value). Using Settings → Move axes, I notice I can't move the X axis any further left than 1mm.

An alternative demonstration is printing a 250×210×0.2mm cuboid, with several perimeters (I have 5 here), and no skirt:

[image]

```
[0;0] point offset:
X -0.95mm
Y  0.12mm
```

MattBlissett, 2023-10-23:
> This issue has been flagged as stale because it has been open for 60 days with no activity. The issue will be closed in 7 days unless someone removes the "stale" label or adds a comment.

Please deactivate this bot.  I spent a couple of hours providing a repeatable test for this issue, and @wavexx must have wasted significant time on failed prints due to it.

You clearly don't intend to fix the issue — even though the printer isn't capable of printing at the advertised dimensions without a fix or workaround — so at least leave old issues open, so they're more easily found by other users.

wavexx, 2023-10-23:
I'm not a fan of stale bots either. It does help to weed old issues which are reported without followup, but for heavily debugged issues and/or stuff you're actively tracking and I need to go there and keep pinging issues. I've seen not about a dozen relevant issues I was monitoring auto-closed. To be honest, not sure how many of those were actually still issues though, but closing without confirmation always felt bad to me.... Anyway, seems like the only way to prevent this is to assign someone, so assigning myself A.T.M.

MattBlissett, 2024-10-25:
> @MattBlissett Is this still an issue with FW 3.14.0

Yes, it's the same in 3.14.0 (Revo build). 

G-code for a 250mm×50mm×0.2mm cuboid generated with the defaults, and then with the X origin set to -1 as described above, is here:

```
; default G-code                                ; x-origin set to -1 g-code
; printing object Shape-Box id:0 copy 0         ; printing object Shape-Box id:0 copy 0
G1 X249.59 Y209.59 F10800                     | G1 X250.59 Y209.59 F10800
G1 Z.2 F720                                     G1 Z.2 F720
M73 P17 R0                                      M73 P17 R0
M73 Q18 S0                                      M73 Q18 S0
G1 E.8 F2100                                    G1 E.8 F2100
M204 S800                                       M204 S800
;TYPE:External perimeter                        ;TYPE:External perimeter
;WIDTH:0.419999                                 ;WIDTH:0.419999
G1 F1200                                        G1 F1200
G1 X.41 Y209.59 E7.81285                      | G1 X1.41 Y209.59 E7.81285     ***
G1 X.41 Y160.41 E1.542                        | G1 X1.41 Y160.41 E1.542       ***
M73 P44 R0                                      M73 P44 R0
M73 Q45 S0                                    | M73 Q44 S0
G1 X249.59 Y160.41 E7.81285                   | G1 X250.59 Y160.41 E7.81285
M73 P49 R0                                      M73 P49 R0
M73 Q50 S0                                      M73 Q50 S0
G1 X249.59 Y185 E.771                         | G1 X250.59 Y185 E.771
M73 P76 R0                                      M73 P76 R0
M73 Q76 S0                                      M73 Q76 S0
G1 X249.59 Y209.53 E.76912                    | G1 X250.59 Y209.53 E.76912
M73 P79 R0                                      M73 P79 R0
M73 Q79 S0                                      M73 Q79 S0
M204 S1000                                      M204 S1000
; stop printing object Shape-Box id:0 copy 0    ; stop printing object Shape-Box id:0 copy 0
```

I believe for the two lines marked `***` the printer doesn't move to 0.41, but probably 1.00, so the shape is 0.59mm too narrow.

closed 2025-01-01 by github-actions[bot]