prusa3d/Prusa-Firmware #2769: [BUG]When scrolling through the sdcard file list from bottom to top files disappear and jump around

jaapp, 2020-06-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- 3.9.0

**MMU Upgrade** - n/a
**MMU upgrade firmware version** - n/a

**SD card or USB/Octoprint** - sdcard

**Describe the bug**
  When you go into the menu and select "print from SD" and scroll all the way down and then scroll 4 clicks back to the top  the files down scroll down but instead the top file is replaced by a different file.

When I scroll up from:
  > MKS_PLA_Triceratops
     MKS_PLA_Vase
     MKS_PLA_Whistle
     MKS_PRUSA_dual_col

This is the result
  > MKS_PLA_Treefrog
     MKS_PLA_Vase
     MKS_PLA_Whistle
     MKS_PRUSA_dual_col
  
**To Reproduce**
  Scroll to the bottom of the file list in the "print from SD" menu, then scroll 4 clicks upward to move the cursor out of the screen.
  
**Expected behavior**
When I scroll up from:
  > MKS_PLA_Triceratops
     MKS_PLA_Vase
     MKS_PLA_Whistle
     MKS_PRUSA_dual_col

I expect this result:
  > MKS_PLA_Treefrog
     MKS_PLA_Triceratops
     MKS_PLA_Vase
     MKS_PLA_Whistle

**G-code**
  n/a

**Video**
  
[image]

leptun, 2020-06-30:
You might want to give this PR a try: https://github.com/prusa3d/Prusa-Firmware/pull/2405
It fixes most issues that the current SD menu has.

jaapp, 2020-06-30:
Don't take this the wrong way but I'm really not interested in compiling my own firmware and debugging this issue. Taking my time to write and film an extensive bug report is all I'm willing to invest as a (new) Prusa customer.

This happened on a freshly build and completely stock machine. Even the SD Card content is stock. I'm sure I'm not the first customer to encounter this.

gogglespisano, 2020-07-01:
I think I've seen something similar.

Sometimes when I spin the knob a bit quickly to move up/down the list I notice that it's missing an entry.  Then I scroll a bit up, then down, or vice versa and it's listed again.

vintagepc, 2020-07-06:
I can also attest to having seen this.

brettvitaz, 2020-07-19:
I no longer click on a file to print until I've scrolled up/down a few times to make sure the display has settled. I have started the wrong file too many times.

gudnimg, 2023-07-03:
I suspect this issue has been resolved now. I have never seen this issue on my MK3S+.

Prusa-Support, 2023-09-20:
It seems safe to close this issue as obsolete but please feel free to submit further feedback.

Michele Moramarco
Prusa Research

closed 2023-09-20 by Prusa-Support