prusa3d/Prusa-Firmware #2770: [BUG]<M221 T0 S95 command doesn't change a flow multiplier value>

Ruhkukah, 2020-07-01:
**Printer type** - [MK3S]
**Printer firmware version**-  [3.9.0]
**SD card

**Describe the bug**
  When using ideaMaker to slice a model, it adds in the very beginning a G-Code "M221 T0 Sxx.00", where xx corresponds to a flow multiplier value, which in most cases on Prusa in equal 95. The bug is that this code doesn't work, retaining the previous flow multiplier value stored in the Prusa's memory because Prusa's firmware incorrectly treats the T0 flag. This happens because of an error in Marlin_main.cpp code:
1. In the code there are two variable "extruder_multiply[№ of extruder]" and "extruder_multiplier[№ of extruder]", which serve similar purpose - they store values of flow multipliers for each extruder. When a command M221 is executed, "Extruder_multiply" is assigned a flow multiplier value but it is never used again in the code, hence the flow multiplier doesn't change after the command M221 is executed.

This problem doesn't happen when a command "M221 Sxx.00" is executed because it follows a different logic in the code.
.   
**To Reproduce**
Add after the startup g-code a command "M221 T0 S90" and see if it changes a flow multiplier value from the Prusa's menu
  
** How to fix** 
Please change the functions **calculate_extruder_multipliers()** and **calculate_extruder_multiplier()** to ensure that T0 argument works as it should.

JohnnyDeer, 2020-07-07:
Hi @Ruhkukah, thank you for reproting.

wavexx, 2020-07-24:
Does IdeaMaker only add a single M221 command by default?
Do you know where is it configurable for the user? What's the intended purpose?

I have zero experience with IdeaMaker, which is why I'm asking.

I've been looking at usages of M221 and I'm interested in knowing how people/software is using it as it affects the dynamics of pressure calculation in the firmware.

Ruhkukah, 2020-07-31:
Hi @wavexx, 

As far as I can see ideaMaker adds a single M221 T0 S##.## command in the beginning G-code. You can change a flow rate value that will directly affect S## value in ideaMaker. The intended purpose is the same as changing flow rate in firmware, which is now fixed in PrusaSlicer to 95%. AFAIK, ideaMaker doesn't use extrusion multiplier, instead it adjusts extrusion rate through M221 command. It is less flexible on Prusa's firmware since a floating point number of S##.## is converted into the integer S##.00 value by Prusa/Marlin's firmware, so you can't adjust the extrusion rate with the same precision as with extrusion multiplier (see "int tmp_code = code_value();" in Marlin_main.cpp).

I can see in Marlin_main.cpp two usage scenarios for extrusion multipliers:
1. Change extrusion rate by the command "destination[i] *= emult;" where "emult" is assigned the value of "extruder_multiplier[]"
2. Adjust firmware retraction rate by using "extrudemultiply" value (not sure that I understood this algorithm correctly), which is not normally used by PrusaSlicer or ideaMaker since all retractions are normally handled through E-#.## code.

Regarding your comment on pressure calculation, in the firmware I don't see how it affects it besides the the command above ("destination[i] *= emult;"), which just changes a number of microsteps for the E axis in a linear fashion. So nothing fancy.

Is it helpful?

wavexx, 2020-08-03:
In PrusaSlicer and slic3r there are 3 ways to control this: directly via filament diameter, extrusion multiplier and indirectly using flow via M221 (currently set to 95% in MK3 variants).

EM and diameter result in different extrusion lengths, while M221 moves the burden to the FW.

Does idea maker allow to tune the diameter? By reading your comment, it doesn't have something extra like Extrusion Multiplier that actually changes the E distances without using M221?

(do you have a screenshot? I'm curious)

closed 2023-10-31 by github-actions[bot]