prusa3d/Prusa-Firmware #2773: [BUG]

brettvitaz, 2020-07-07:
**Printer type**
MK3S

**Printer firmware version**
3.9.0-3421

**MMU Upgrade**
MMU2S

**MMU upgrade firmware version** 
1.0.6

**SD card or USB/Octoprint**
Print from SD Card, Octoprint is connected

**Describe the bug**
Occasionally after printing has appeared to complete, the printer head will move to the front right, unload the filament and prompt to load new filament. Then it will move to the rear left as if to complete, then it will start moving toward the center and display "Inactivity Timeout" on the screen. The printer must be reset to continue.
  
**To Reproduce**
Printing certain gcode files causes this more often, but I have not made an effort to recreate error due to time and filament required.
  
**Expected behavior**
Print to complete without unloading filament, and without needing reset.
  
**G-code**
[Sensor_Head_0.2mm_PLA_MK3SMMU2S_1h42m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4887471/Sensor_Head_0.2mm_PLA_MK3SMMU2S_1h42m.gcode.zip)

**Video**
_I will take this if necessary._

**Octoprint terminal output**
```
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: unload
Recv: mmu_get_response() returning: 0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: USART2 rx Full!!!
Recv: USART2 rx Full!!!
Recv: USART2 rx Full!!!
Recv: USART2 rx Full!!!
[... 238 lines cut ...]
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: DoDone printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Done printing file
Recv: echo:17 hours 2 minutes
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Finishing" to "Error: Printer halted. kill() called!"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor
```

awenelo, 2020-07-08:
Can you try slicing it in PrusaSlicer 2.2.0?
 I think that it might have something with the text at the end of the file(probably generated by a bug), which the printer doesn't appear to like(starts at `Recv: echo:Unknown command: ".01170"(2)`). By loading the Gcode into Cura's Gcode viewer, it appears that the move to the center is caused by this section (loading it in without this section results in a normal final nozzle position, with this section results in a nozzle position above the center of the print) Also, before this section happens, `M104 S0` and `M140 S0` are executed which turn off the nozzle and bed, but this section tells the printer to extrude, indicating further that this text is not intentional.
```
.01170
G1 X96.757 Y141.706 E0.01181
G1 X96.935 Y141.775 E0.00496
G1 X106.895 Y141.775 E0.25952
G1 X113.107 Y139.446 E0.17287
G1 X112.896 Y139.300 E0.00670
G1 X111.999 Y138.464 E0.03195
G1 X111.551 Y138.201 E0.01354
G1 X111.103 Y138.075 E0.01213
G1 X110.654 Y138.053 E0.01170
G1 X110.206 Y138.120 E0.01181
G1 X109.758 Y138.293 E0.01252
G1 X109.309 Y138.601 E0.01418
G1 X108.413 Y139.437 E0.03195
G1 X107.965 Y139.700 E0.01354
G1 X107.516 Y139.826 E0.01213
G1 X
```

brettvitaz, 2020-07-08:
I will try that. I am now also inclined to blame the Toshiba FlashAir SD card since I have seen on a few occasions that a copied gcode file would report as "incomplete". I pulled the file from the SD card that I posted here.

brettvitaz, 2020-07-10:
It's pretty obvious that the gcode was corrupt. Closing

closed 2020-07-10 by brettvitaz