prusa3d/Prusa-Firmware #2777: [BUG] FW 3.9.0 - Backlight doesn't turn to bright mode when interacting after filament runout and cooldown

devonmerner, 2020-07-11:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - None

**SD card or USB/Octoprint** SD card

**Describe the bug**
When the filament sensor trips mid-print and the hot-end cools down after a timeout, the screen will stay in dim mode until the hot-end reaches full temperature even though the user had to press the knob to start the unload procedure.

I have my backlight dim mode set to 0 (off) with a 120 second timeout.

With my dim brightness set to 0, this made it very difficult to tell whether the printer was heating up or if it was stuck/off.
  
**To Reproduce**
1. Set backlight dim setting to something low (or 0).  Set your backlight timeout set to something low for testing.
2. Start a print via SD card
3. Cut the filament strand early to trip the filament runout sensor
4. Wait until the screen dims and the hot-end cools down (the extruder fan will stop)
5. Press the knob on the printer to begin the unload process

The printer should start to warm up the hot-end but the screen will stay dimmed/off until it reaches full temperature.
  
**Expected behavior**
That the screen will light up immediately when a user interacts with the printer after a filament runout (and hot-end cooldown timeout) to show the temperature of the hot-end as it heats back up.

leptun, 2020-07-11:
Thanks for the feedback. I’ll look into this.

devonmerner, 2020-07-11:
Just tested on my second MK3S which needed a new spool mid print.  Same issue.

On my second printer, the filament didn't unload correctly and I had to select "No" when asked if it unloaded correctly.  After it locked the X-stepper motor and I cleaned out the old filament and prompted the printer to continue, it seems like the display was forced back into dim mode despite me having just pressing the knob on it (and the backlight timeout still being set to 120 seconds).

It seems like there might be a few edge-cases during the filament runout procedure regarding the backlight settings that should be investigated.

DesC21, 2022-02-15:
sorry for bringing this old issue back up but I also have this problem, as well as two other scenarios were the display stays in dim mode and doesn't turn on again. This happens for me when I "load filament to nozzle" and the question about the correct color is displayed. As well as the filament selection after starting a print in MMU2S single mode. This has been the case from when the display dimming has been introduced, it never changed (I am running an MK3S with 3.10.1 and MMU2S with 1.0.6).

leptun, 2022-02-15:
I'll mark this for 3.12 so we don't forget about it... again... Sorry for saying that I'll look into it and forgetting about it.

lukasradek, 2022-04-04:
Looking forward to this fix. Can confirm it happens on every "confirmation popup".

PR #4010 (2023-02-21): backlight: always allow backlight to dim/wake
This commit adds the ability for the firmware to dim and wake the backlight when LCD updates are disabled. Such as in the MMU error screen or when rendering full screen messages which typically disable the LCD updates to prevent the status screen from rendering.

Fixes #2777

Fixes an issue where the backlight may stay dim on MMU error screen even though the knob is pressed or rotated. Only happens when the Sound settings are set to Silent.

Change in memory:
Flash: -6 bytes
SRAM: +1 byte
commit db878c99 (2023-02-21): backlight: always allow backlight to dim/wake

This commit adds the ability for the firmware to dim and wake the
backlight when LCD updates are disabled. Such as in the MMU error screen
or when rendering full screen messages which typically
disable the LCD updates to prevent the status screen from rendering.

Fixes #2777

Change in memory:
Flash: -26 bytes
SRAM: +1 byte
closed 2023-02-21 by 3d-gussner