prusa3d/Prusa-Firmware #2778: [ENHANCEMENT] Improve Z Leveling method

fscheps, 2020-07-17:
**MK3S**

Improving Z  Leveling method:
As a new user to 3D Printing I found that one of the most difficult things to get right is the Z Levelling. And because this is directly related to the quality of the first layer print, I think its quite important that as manufacturers of the printer you try to make it as easy as possible for new joiners to the 3D Printing ecosystem. 

The GCode included on [this guide](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-assembly-and-first-prints-troubleshooting/life-adjust-z-my-way/), I think its very useful and should be included on the SD card, also should be referenced to be used as a more realistic method to do a proper Z Leveling.  The method that comes on the Firmware is too short to do a proper adjustment. Especially when its not clear what is an acceptable outcome. 

I think that if in the future the Z Leveling could be completely automated, this will be a game changer. People will stop wasting so much time (and filament) to get this right and print the right way.

Thank you for your consideration.

wavexx, 2020-08-17:
I read this, but the subject is somewhat misleading :)

What you're suggesting exactly? Changing the built-in first layer calibration pattern, give more precise instructions?

There's no perfect "Z height" IMHO. It depends what material you're going to use, how much squish/adhesion you want (for PETG on PEI you'd likely want a little less), if you're using a textured sheet or not... and it also goes to taste.

The acceptable range for PLA on the smooth PEI sheets is surprisingly broad too.

fscheps, 2020-08-17:
Hi there, 

Well you seem to be acknowledging some issues of the Z Levelling management :) 

1 - The pattern that is included on the firmware is not very useful as it's too short and blocked by the device itself. The square pattern as suggested on the guide I linked on my initial post would be more helpful, a lot of people would waste way less time on this.

2 - Another thing that could be done is to save different layer heights for different materials and different sheets. Because in the end if we have 2 variables: Material and sheet that would change the Z Level, then why not allowing us to save these variables, so we don't have to re-set the Z level each time we change material and sheet?

I am quite new to all this 3D Printing world and by far this Z Levelling is the most time-consuming and complex part to get it right.

Cheers

Fernando

wavexx, 2020-08-17:
In FW 3.9 this is partially addresses as you can now have predefined offsets saved as "Sheets". Essentially it allows to configure the perfect distance to each sheet, but that could also be used to save a sheet/material combination as well (the names are arbitrary and you can redefine them). I personally wished the "material" component would be done via gcode (essentially having a Z offset in the material profile instead of having it in the machine settings).

The built-in pattern is IMHO "good enough" to get a good starting point for the sheet. I've seen several people looking from above/following the extruder to gauge the squish, but I always look from the side/front to see the distance from the nozzle/bed by eye instead.

I always fine-tune the distance with live-z adjustment on the first print.

paulcdb, 2020-09-30:
> People will stop wasting so much time (and filament) to get this right and print the right way.

Do first calibration layer, once complete, measure the box with calipers, adjust if needed, done.

People seem to spend far too long trying to achieve something more instead of enjoying what is a great hobby! If anything, the advice could be changed for those new to 3D printing to measure the box with calipers, but at the end of the day, it's not really needed and Prusa has the best system for leveling right now. You just need to learn what is OK and what isn't and that only comes when you stop trying to get the 'perfect' level and go print stuff!

closed 2023-10-31 by github-actions[bot]