prusa3d/Prusa-Firmware #2779: [FEATURE REQUEST] User voice like page to vote up and down for new features

fscheps, 2020-07-17:
I would like to propose you create a User Voice type of page, where every user from the 3D Printing community can vote up for new stuff. This way you will be able to better understand what the people are asking for to get on their printers, as enhancements, etc.
This could be a way: 
https://www.uservoice.com/product

But there are Wordpress Plugins that allow you to do the same for example: 
https://themeforest.net/item/discy-social-questions-and-answers-wordpress-theme/19281265

Thank you for your consideration

draeath, 2020-07-18:
I concur, there should be a better way for users to submit feedback than dog-piling in PR comments, such as in #2692

mikolaszuza, 2020-08-04:
I like that idea :)

closed 2023-10-31 by github-actions[bot]