prusa3d/Prusa-Firmware #2780: [ENHANCEMENT] Improvement on option to finish one object at a time

fscheps, 2020-07-17:
Prusa MK3S

When I place multiple small objects on the bed so they can be printed all at the time time. It would be great if we would get more information than just this error message:
https://www.dropbox.com/s/kpqgyn8hwubynf6/Error%20Prusa%20Slicer.png?dl=0

Would be great if slicer could show us which objects are interfering with each other so we can move them and place them in a better place on the print bed. 

Thank you for your consideration

leptun, 2020-07-17:
This is a PrusaSlicer related feature request. Please post the request here: https://github.com/prusa3d/PrusaSlicer/issues

fscheps, 2020-07-17:
Done, thanks

awenelo, 2020-07-17:
@fscheps, if this is for PrusaSlicer, can you close this issue please?

closed 2020-07-17 by fscheps