prusa3d/Prusa-Firmware #2782: [BUG] Wrong calculated cumulative print time

CaveRenderPro, 2020-07-19:
The cumulative print time is much higher than the sum of all individual print times. After a print of 33 minutes the cumulative print time "jumped" from 59d 23h to 73d 22h. 

Maybe this occures when a print is aborted by the user.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0

CaveRenderPro, 2020-07-25:
This issue is looks simular: [https://github.com/prusa3d/Prusa-Firmware/issues/2745](https://github.com/prusa3d/Prusa-Firmware/issues/2745)

crabbedh, 2022-03-23:
Yes.. I am seeing that my print time statistic is **stuck** at 6h 50min... It was this value a day ago and I just completed an 11 hr print job and the print job completed properly. I have 3.10.1-4697  MK3S+

3d-gussner, 2022-03-23:
> Yes.. I am seeing that my print time statistic is **stuck** at 6h 50min... It was this value a day ago and I just completed an 11 hr print job and the print job completed properly. I have 3.10.1-4697  MK3S+

Did you print from SD or via a host like Octoprint? At this moment host prints don't get into the stats. It is planned for one of the next releases.

crabbedh, 2022-03-23:
Ahh.. never even thought of this.. yes.. I have a Pi4 and print exclusively through that. Then I am assuming the factory reset does not zero the print stats.. Yes.. the print hours should track hours the extruder is working no matter how it is fired.. from SD, manually or via gcode from the comm port..   Same with filament length used.

3d-gussner, 2022-03-23:
The factory reset, resets everything also the stats.

crabbedh, 2022-03-24:
okay.. then the factory reset needs to be harder to activate.. it is too similar to what one does to raise the Z axis after a crash or if user has stopped print by hitting reset button.. A yes/no question on factory reset would be sufficient.

closed 2023-10-31 by github-actions[bot]