prusa3d/Prusa-Firmware #2783: [ENHANCEMENT] Allow Z to find it's own maximum height regardless of version/mods instead of =210mm

yoyo42, 2020-07-19:
I have an i3 mk3s where the maximum Z travel should be a touch over 217mm. The firmware forces a hard stop at 210mm.

When calibrating Z it runs right to the stops, so the printer always knows how much Z height it has - please allow them to use it.

Some mods allow a lot more height than that, well over 300mm. Users should not need to recompile firmware to use this extra height - making it a switchable or settable limit on a menu would be a good compromise.

wavexx, 2020-07-24:
That would be nice to have.

The only downsize I see is that XYZ cal also has a limit to how much to extend the Z move to avoid ramming the axis if stallguard doesn't work as expected.

There are probably some other special cases where the distance has been defined as static for the same reason.

As you using a modded extruder to get the extra height, or is it stock?
In such cases you should always be able to change the variant files and rebuild the FW.

Panayiotis-git, 2020-07-25:
I think that the safety measures should include user responses in case of abnormal events. If for example the ramming of the z-axis doesn't occur as was expected, a questionaire to the user like:

a. the z-axis hasn't moved (=> error?)
b. the z-axis has more to go (=> raise some more)
c. the z-axis has in fact moved all the way up (=> error for not detecting the ramming? Could also change the sensitivity.)

would provide the necessary feedback to the procedure. The result should be the detection of the z height. Of course a final verification, in case of a longer height than the expected one was detected, could be scheduled at the end of the procedure.

On the other hand, the number of questions should be minimal, just what is needed to finish the procedure.

yoyo42, 2020-07-25:
Mine is only a Mk3 modded to a Mk3S so my Z is only 217mm but I actually needed all of that the other day, that's what prompted this.

Some mods apparently go well over 300mm though. I know the firmware can be recompiled to change pretty much anything but it shouldn't really be needed to get the best from a stock printer at least, and while changing that it can easily be made to handle a generic extension.

wavexx, 2020-07-29:
This *definitely* doesn't need to happen on stock ;)

paulcdb, 2020-09-30:
> This _definitely_ doesn't need to happen on stock ;)

Yes it does!

I needed to print something 212mm high as part of multi section item and the hard limit meant the previous parts already printed would've gone to waste if I couldn't get that part printed. Luckily I manged to get that part printed but I'm sure there's plenty of reasons that slight extension can make a huge difference.

Since Prusa now have a service menu, it could go there with a warning beforehand any changes are a user responsibility. You do something stupid, it's on you but I don't see how anyone can damage a printer this way that would risk injury in the first place!

Don't deny a features just because some idiot might set it to 300 and then whine it can't print 300mm high.

DRracer, 2020-09-30:
@yoyo42 @paulcdb This is exactly the right case to compile your own firmware - it is open source, so you can tweak your machine beyond factory-verified limits. We are not planning to add tweaking options for modified machines.

yoyo42, 2020-10-05:
Does a Mk3 upgraded to a Mk3S count as 'modified'? Couldn't it just read the height found during the Z calibration and stop at that, whatever it may be? I've never been a fan of arbitrary limits.

closed 2023-10-31 by github-actions[bot]