prusa3d/Prusa-Firmware #2787: MINTEMP functionality is just wrong

Resonanz, 2020-08-01:
MK3S

My printer sits in a thermally insulated box in my garage. It is cold at the mo (~10 degC) and the printer won't start and gives a MINTEMP error. 

Thing is, the heated bed will heat the thermal box IF the thermal bed was actually able to switch on. But the firmware stops it from switching on at all.

The solution is to get a hot air gun and heat the thermal sensor, thus enabling the printer to start, from which point it runs perfectly.

awenelo, 2020-08-02:
I don't see the problem with this. We stop the bed from turning on because we are reading values that could mean that the temperature sensors are disconnected. This is to prevent us from heating at full power forever (as all the information provided indicates that temperature readings are incorrect), which would cause a fire.

Resonanz, 2020-08-02:
> I don't see the problem with this. We stop the bed from turning on because we are reading values that could mean that the temperature sensors are disconnected. This is to prevent us from heating at [...]

Well, I think you should see a problem with this. If your car was parked in the garage and the temperature was 10°C, and your car refused to start, you would think your car is faulty. That is what I think about my printer. The software is faulty.￼ my printer should not stop working just because the temperature is a few degrees lower than a somewhat arbitrary value that has been chosen by somebody.

I am a hardware and software engineer, ￼so I am quite aware of how to measure temperature using both hardware and software.￼￼ I don't know how PRUSA have designed the temperature sensor in my printer however the way it is done at present is evidently a poor design if it cannot distinguish between being disconnected or having a broken wire and being perfectly healthy and functional.

But all of that aside, is there a potential fix such as allowing the user to set the minimum temperature, or asking the user via the LCD display to approve the operation of the printer by pressing the button.￼￼￼￼ 

Thanks.

wavexx, 2020-08-03:
@Resonanz can you post all the temperatures you see reported in Support -> Temperatures when attempting to start the printer from cold?

Does the printer report MINTEMP / MINTEMP BED immediately and/or does it take longer?

Was it done on FW 3.9.0?

Checking for disconnection is not the only reason MINTEMP is there. The thermistors and/or heaters can fail in a number of ways and it's much better to be over-conservative instead of keeping the heat on. That being said FW 3.9.0 should still allow the printer to turn on in cold environments, allowing a delay before the temperatures start to climb.

3d-gussner, 2020-08-04:
Internal tests have been done and 10°C is considered a "safe" temp to be read, everything below get's more "guessing" if the sensor is working correctly or not.

But I can see the point @Resonanz has, it would be helpful to have a "user action" menu to force the printer to heat up.
If this should be considered as an option:
- User should acknowledge every xx seconds on LCD that he is watching the printer during heat up
  - this should stop as soon the printer can read reasonable values during heat up
- Remote printing via USB is a different thing as the user "normally" isn't there to confirm
  - Not sure if we should allow one or two short heat up sequences to see if the printer get's reasonable values to make this an auto heat up feature.

**No one want's a printer that may get on fire.**

There is a reason for MIN- and MAXTEMP in the firmware.

DRracer, 2020-09-25:
@Resonanz The printer has well defined and tested operational limits. Your comparison with a car is not fair - a car is designed to start/operate within a wider range of temperatures.
The primary concern in printer's design is safety and therefore there are features which prevent the ordinary user from operating the machine beyond its limits.
If the limits do not work for your application, you have the possibility tweak it - both the HW and the SW are open source - but obviously you cannot blame the vendor in case something goes wrong.

Therefore I'm closing this issue.

closed 2020-09-25 by DRracer

robin-gdwl, 2023-04-02:
I find the constant failure to acknowledge this issue as a problem to be incredibly frustrating. 
Of course there are reasons why there are mintemp warnings and watchdogs, however there are many use cases where it is beneficial to have the option to use a heating element to get a printer up to working temperature. 
To even include a thermistor that is so unreliable below 10°C in a 1000€ Printer is not acceptable in my opinion. 
The solution is of course not to simply deactivate the thermistor watchdog alltogether. Instead there needs to be a 10s grace-period in which the printer is allowed to heat and only enter an error state if no temperature change is registered. This is the way many printers and other machines are doing it. The heating element is not powerful enough to start a fire in 10s especially when the printer had been switched off for a period of time. Other solutions such as ambient temperature sensors or multiple thermistors could also increase safety while allowing for this edge-case. 
This printer is not just a Hobby machine but also an industrial tool and having a slightly cold workshop should be an expected and manageable circumstance that should not need unsafe, untested custom firmware changes.

Resonanz, 2023-04-02:
Well, the answer is simple. I will not be purchasing another Prusa printer. Period.

Prusa-Support, 2023-05-12:
Thanks for your feedback.
We were reconsidering this aspect and discussed it very extensively in a later issue.

This not only may feel unsafe but it's hardly doable.
Read more on the issue https://github.com/prusa3d/Prusa-Firmware/issues/3053.

In this specific scenario, an external heater could do the trick.

Michele Moramarco
Prusa Research

rvalle, 2023-10-26:
@Prusa-Support there are many issues on MIN-TEMP, implementing a MIN-TEMP disable setting is the best option.

The printer is also designed for using FANs, and there is a FAN check disablement.

In some situations the MIN-TEMP functionality is simply wrong, and needs to be disabled.

Prusa-Support, 2023-11-29:
Please check https://github.com/prusa3d/Prusa-Firmware/issues/3053 and consider taking part in the discussion there.
That thread is quite extensive, and you can find further reasoning about the MINTEMP error and the safety measures.

Michele Moramarco
Prusa Research