prusa3d/Prusa-Firmware #2795: [BUG]<Enter comprehensive title>

bszonye, 2020-08-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI
**Printer firmware version**-  4.2.0

**Describe the bug**
The Prusa MINI can beep, but I cannot produce a beep myself with the M300 code.
  
**To Reproduce**
I attempted to play default beeps and a simple tune using some gcode intended for an end code script. The script successfully homes and parks the print head, but it does not produce any tones from the speaker.
  
**Expected behavior**
I expected the script below to home the print head, beep, play the first phrase of Beethoven's 5th, beep again, and then park the print head.
  
**G-code**
```
G28 W ; home without level

M300 ; beep

; 555ms per beat
M300 S392 P120 ; G4 1/8
M300 S0 P20
M300 S392 P120 ; G4 1/8
M300 S0 P20
M300 S392 P120 ; G4 1/8
M300 S0 P20
M300 S311 P600 ; Eb4 1/2 fermata                                                                                                                                 

M300 ; beep

G1 Z53.25 F720 ; Move print head up
G1 X178 Y180 F4200 ; park print head
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow
M84 ; disable motors
M73 P100 R0
```

leptun, 2020-08-08:
This issue tracker is for the 8 bit firmware only. For issues regarding the Prusa Mini firmware, please post here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

bszonye, 2020-08-08:
Sorry about that! I see that there is an open issue for this over there: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/334

closed 2020-08-08 by bszonye