prusa3d/Prusa-Firmware #2797: [ENHANCEMENT] MMU2S add delay after IR sensor initial load trigger

ANTALIFE, 2020-08-09:
Printer:
**Prusa i3 MK3S + MMU2S**

**Is your enhancement related to a problem? Please describe.**
I find that once or so during a long MMU2S print (>15hrs) I run into a loading issue. It turns out this is due to the filament tip having an irregular diameter, which causes the IR sensor (at hotend) to suddenly think that no filament is present (even though there is good filament after the irregularity) and initiate an unload. Sometimes the unload fails as the extruder motor catches the on initial tip and "blocks up"

[image]

A couple of other key things is that I am:
- Run a **[modded printer](https://www.antalife.com/2020/07/project-prusa-i3-mk3s-mmu2s.html)**
- Using **[PrusaSlicer 2.2.0 DRIBBLING by antimix](https://github.com/antimix/PrusaSlicer/releases)**

For more info see post of Prusa forums:
**https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/where-to-add-a-delay-during-load-in-firmware**

**Describe the solution you'd like**
I want to add a slight delay after the IR sensor (at hotend) confirm filament load, this way the filament diameter irregularity will be masked. From my understanding I would need to do this in "mmu.cpp"
Also I know that I can try playing around in PrusaSlicer settings for each filament, but I still think I will run into this issue with new filaments. Hence I was hoping to make the firmware more robust against this type of fault

3d-gussner, 2020-08-10:
I don't think that Prusa will change the code because of this strange filament tip shape.
But that is also not what you are asking for you want to know where you can modify the firmware to make your own improved version to handle these shapes.

Checkout https://github.com/BondtechAB/Bondtech-Prusa-Firmware/tree/Bondtech-MK3-FW390 as they have sightly different distances in their extruder compared to Prusa R4/R5 version and extrude some extra filament.

You could also try to play with https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L360-L381 values to change the behavior of the printer during the MMU2s filament change.

ANTALIFE, 2020-08-10:
> I don't think that Prusa will change the code because of this strange filament tip shape.
> But that is also not what you are asking for you want to know where you can modify the firmware to make your [...]

Thanks again, will have a look at above

3d-gussner, 2020-08-11:
Give it a try and let us know when you find something.
Still think that this tip looks very strange and you should investigate why it is like that.

ANTALIFE, 2020-08-17:
Update time

Have modified 3.9.0 FW to better handle diameter irregularities, see [MK3_3.9.0-ANT](https://github.com/ANTALIFE/Prusa-Firmware/tree/MK3_3.9.0-ANT)

Only change is to ["can_load()" function of "mmu.cpp"](https://github.com/prusa3d/Prusa-Firmware/compare/MK3...ANTALIFE:MK3_3.9.0-ANT), as here I adjusted the "filament_detected_count" threshold from 26 (-4) to 5 (-25). Doing so means the IR sensor has to see 5 instances of valid load (vs the old 26) for the filament load to be considered successful. For one of my 5 colour prints above change has reduced failed loads from 10 to 0 :D

RolfHodl, 2020-08-18:
> Update time
>
> Have modified 3.9.0 FW to better handle diameter irregularities, see [MK3_3.9.0-ANT](https://github.com/ANTALIFE/Prusa-Firmware/tree/MK3_3.9.0-ANT)
>
> Only change is to ["can_load()" [...]

Can you compile a version of the actual 3.9.0 please, I think this is not the final 3.9.0?

ANTALIFE, 2020-08-18:
> Can you compile a version of the actual 3.9.0 please, I think this is not the final 3.9.0?

I branched from MK3_3.9.0 which i'm pretty sure is the final 3.9.0?

RolfHodl, 2020-08-18:
Was just irretaded by "This branch is 4 commits ahead, 83 commits behind prusa3d:MK3." when I looked your branch up. Thanks - I will try your firmware and report back.

closed 2023-10-31 by github-actions[bot]