prusa3d/Prusa-Firmware #2798: [BUG] MMU2S Extruder turns wrong way.

mrjackson, 2020-08-09:
**Printer type** - MK3S
**Printer firmware version**-  3.8.1

**MMU Upgrade** -  MMU2S
**MMU upgrade firmware version 1.0.6

**SD card or USB/Octoprint**
  Both

**Describe the bug**
 When filament unloads with the mmu2s, once the IR sensor no longer detects filament the extruder motor stop, the mmu picks up the filament to finish the unload, and the extruder starts turning in the loading direction.
This sometimes results in the extruder holding on to the filament overpowering the mmu and resulting in an unload failure.
  
**To Reproduce**
Print with the mmu.
  
**Expected behavior**
 Extruder should turn in the unload direction while the mmu is unloading.
  
**G-code**
 Happens on all prints.

**Video**

3d-gussner, 2020-08-10:
What changed? Did it work before?

Did you check the MMU2 Tower IR sensor triggering correctly? Did you clean the bondtech gears as debris can false trigger the MMU2 Tower IR sensor?

Do you use stock MK3S+MMU2S or do you have any mods like Bondtech extruder?

Do you use original Prusa firmware or a custom firmware?

mrjackson, 2020-08-10:
As far as I can remember, it's always been this way I've had the MMU2S a little over a year. It doesn't fail often but when it does it's quite annoying to fix.

IR and FINDA sensors are working perfect, gears are clear.

Stock everything, only change is the nozzle.

Stock firmware.

3d-gussner, 2020-08-12:
I can't promise a quick solution but we will try to look into it. Stay tuned.

ANTALIFE, 2020-08-20:
Sounds like your tips don't have an "ideal' shape, causing the tip to get caught in the extruder motor during unload
I also have experienced similar issues (see point 3 of picture [here](https://github.com/prusa3d/Prusa-Firmware/issues/2797))

In the end I had to do some hardware & firmware mods to improve reliability with my MMU2S:
https://www.antalife.com/2020/07/project-prusa-i3-mk3s-mmu2s.html

mrjackson, 2020-08-20:
> Sounds like your tips don't have an "ideal' shape, causing the tip to get caught in the extruder motor during unload
> I also have experienced similar issues (see point 3 of picture [...]

No, I don't need to do any of these things. I have printed many many hours with very few interventions required. Last week I printed 2 of the smiling owls you have in your blog with 0 interventions.
It's rare that this issue causes me problems, but it should be a fairly straight forward fix and could help reliability for anyone that does have some other issues. And with the cutting being enabled it could help even more.

ANTALIFE, 2020-08-20:
> G-code
Happens on all prints.

> It's rare that this issue causes me problems

( ͡° ͜ʖ ͡°)

mrjackson, 2020-08-20:
> 
> 
> > G-code
> > Happens on all prints.
> 
> > It's rare that this issue causes me problems
> 
> ( ͡° ͜ʖ ͡°)

The extruder rotating backwards on unload happens on all prints, rarely does it cause me an issue.

mrjackson, 2021-09-22:
Any update? It's been a year.

mrjackson, 2022-10-17:
And now 2 years. Maybe with the reworked MMU firmware?

Prusa-Support, 2022-10-27:
Thank you for reporting this.

The bug is confirmed, but, sadly, the firmware aspects concerning the MMU2S have not been updated much in the last two years. Actually, the MMU2S-specific firmware was not updated this far because our developers are working on a big firmware overhaul.

As you may know, this particular bug should not be bothering very often: normally the filament should not find itself between the extruder gears when the buggy behavior shows up. In other words, this is probably not a priority fix, but our developers are willing to fix it.
The issue will stay open until then.

Michele Moramarco
Prusa Research

gudnimg, 2023-07-03:
Adding this to the 3.13.0 milestone. I have not seen this issue with the new MMU FW.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg