prusa3d/Prusa-Firmware #2799: [BUG] MK3S 3.9.0 compiled firmware does not match Prusa release

ANTALIFE, 2020-08-12:
**OS** Fedora 5.7.14-200

**Describe the bug**
Compiled firmware from 3.9.0 git branch (MK3_3.9.0) does not match Prusa release 

I tried compiling on Windows 10 but had issues like [other people](https://github.com/prusa3d/Prusa-Firmware/issues/2337). So I installed Fedora on my laptop and followed the [Linux compile instructions](https://github.com/prusa3d/Prusa-Firmware) from Prusa. I seem to compile without any issues, but when I compare the size (and contents) of my hex file to the one from [Prusa](https://www.prusa3d.com/drivers/) the two are different (see below for files). Could anyone tell me how this can be?

**Sample files**
https://drive.google.com/drive/folders/1QXONciumbPcrTrNbyJ1rokzDtNl7zqIO?usp=sharing
  
**Expected behavior**
Given that I setup the environment correctly then there should be no difference between the two hex files ?_?

3d-gussner, 2020-08-12:
Did you try to compare the two hex files with winmerge or other tools?
You will never get the exact same hex file  twice as during compiling the date/time are included, also I don't know how Prusa build server adds at the end of the hex some data.
I couldn't get an identical hex file but got with WLS and Ubuntu VM very close except the time stamp and the last line.

3d-gussner, 2020-08-12:
Also very important is to know to use the source from the release to compile if you want to compare your version against the release hex files.
The MK3 branch is the development branch, so it can be very fast different from the release tag version.

3d-gussner, 2020-08-12:
This is not a bug😉 it is more a question.

ANTALIFE, 2020-08-12:
> Did you try to compare the two hex files with winmerge or other tools?
> You will never get the exact same hex file twice as during compiling the date/time are included, also I don't know how Prusa [...]

Yuppo, compared using Beyond Compare and saw lots of red D:
Also did see that Prusa hex file has the config name at the end, do you know where the time & date stamp is in the hex file?

===================================================================================

> Also very important is to know to use the source from the release to compile if you want to compare your version against the release hex files.
> The MK3 branch is the development branch, so it can be very fast different from the release tag version.

Makes sense, that's why I changed to MK3_3.9.0 branch as I assumed 3.9.0 was compiled here?

===================================================================================

> This is not a bug😉 it is more a question.

Ah my bad, is it too late to change the label?

3d-gussner, 2020-08-12:
As said use the source.zip from the release or the Tag in git checkout v3.9.0

https://github.com/prusa3d/Prusa-Firmware/tree/v3.9.0

Can not say where exactly the timestamp is located, but if you have a hex viewer you will find it.

3d-gussner, 2020-08-12:
Just done a self compiled hex file and compared it to release hex file.

[image]

Self compiled with "source.zip" from release
```
:1072C000553253004E4F4D4B3353002E01417567DD
:1072D000203132203230323000436F6D70696C657E
:1072E000643A2000286E6F6E652C206465666175B7
:1072F0006C7420636F6E6669672900417567203181
:107300003220323032302031383A33313A333100A2
```
vs release hex
```
:1072C000553253004E4F4D4B3353002E014D6179D3
:1072D000203138203230323000436F6D70696C6578
:1072E000643A2000286E6F6E652C206465666175B7
:1072F0006C7420636F6E66696729004D6179203177
:107300003820323032302031373A31313A3135009D
```
using https://www.rapidtables.com/convert/number/hex-to-ascii.html
You can see the diff:
Self compiled
```
rÀU2SNOMK3S.AugÝrÐ 12 2020Compile~ràd: (none, defau·rðlt config)Aug 1s2 2020 18:31:31¢
```
vs release hex
```
rÀU2SNOMK3S.MayÓrÐ 18 2020Compilexràd: (none, defau·rðlt config)May 1ws8 2020 17:11:15
```

ANTALIFE, 2020-08-12:
Ah was I not selecting the branch correctly... Thanks for above, will try it after work

ANTALIFE, 2020-08-13:
Tried with above repo: 
https://github.com/prusa3d/Prusa-Firmware/tree/v3.9.0

Still get a huge difference between what I compile and what you guys compile D: 
Again I don't see any unexpected errors during compile (see "Fedora-terminal.txt"), but there seems way too much different. Plus when I upload my FW ("My-firmware.hex") to printer a few options become disabled, like spool join which shows as N/A
https://drive.google.com/drive/folders/1TPdhWBeQHGp1Y5BMv2ibwR_NSoT-6x5e?usp=sharing
[image]

Am I still missing something or is this just a reality of compiling on different systems?

3d-gussner, 2020-08-13:
That is quite different.
Did you use PF-build.sh with all languages and Gold to compile the firmware?
As you can see my fresh compiled hex file just differs at timestamps and the last line with the variant

3d-gussner, 2020-08-13:
Hope you compare unmodified self compiled firmware to the release hex.
Changed source code will generate changed hex.

ANTALIFE, 2020-08-13:
To build it I used the following on the 3.9.0 branch [you linked](https://github.com/prusa3d/Prusa-Firmware/tree/v3.9.0)
[AYYLIEN@localhost Prusa-Firmware-3.9.0]$ **./build.sh** 

What do you mean with "PF-build.sh" & "Gold"? 

Also yuppo am doing this on unmodified file. Want to make sure my environment is configured correctly before I start changing stuff

ANTALIFE, 2020-08-13:
Just tried compiling in Ubuntu 20.04.1 (Live USB) and same issue, stuff is way too different D:
[image]

And like before here are all the files:
https://drive.google.com/drive/folders/1EFh6PgwtMQ9FLSyF_JYOk34wuV-hgHBw?usp=sharing

3d-gussner, 2020-08-13:
Please start ". /PF-build.sh" it will ask you to install few things if they are missing and then download a known build environment. Then you can choose the variant, if it should buil the all languages version and as it does know if that is a final source code choose Gold.
This is what I have done.

ANTALIFE, 2020-08-13:
Thanks heaps @3d-gussner, it's looking much better now!
[image]

Looks like the culprit is using "./build.sh" instead of ". /PF-build.sh"
Might be a good idea to update the github instructions for Linux?
https://github.com/prusa3d/Prusa-Firmware

ANTALIFE, 2020-08-13:
Hmm, even if I do a proper clone of git repo "git clone https://github.com/prusa3d/Prusa-Firmware.git" and change to MK3_3.9.0 branch "git switch MK3_3.9.0", I still get similar differneces as before:

> :1072C000553253004E4F4D4B3353002E01417567DD
:1072D000203133203230323000436F6D70696C657D
:1072F0006C7420636F6E6669672900417567203181
:107300003320323032302032313A31333A3539009D
:109450002E2073656E736F7220762E3A0041756709
:109460002031332032303230004533447636667551
:10966000008813000010101020FFFFFFFF4D37305F
:10000000A55AB44BF6256D01000073636D01000025
:1025F000656E69205A00A55AB44BB62B6D010000D8
:1026000065646D010000EA023C034D035D03640351
:1051B000852A6D01000073656D010000EA02360367
:107C300000A55AB44B8A2C6D01000072666D0100DC
:107C400000EA023603460356035C0362037503B57C
:10A8C0002B6D01000074696D010000EA023503463A
:10D3C000A55AB44B27286D0100006C706D01000058

Which comes out to 

> rÀU2SNOMK3S.AugÝrÐ 13 2020Compile}rðlt config)Aug 1s3 2020 21:13:59P. sensor v.:Aug	` 13 2020E3Dv6fuQ` ÿÿÿÿM70_¥Z´Kö%mscm%%ðeni Z¥Z´K¶+mØ&edmê<M]dQQ°*msemê6g|0¥Z´K,mrfmÜ|@ê6FV\buµ|¨À+mtimê5F:ÓÀ¥Z´K'(mlpmX

Would it be safe to say this is as close as it gets?

3d-gussner, 2020-08-13:
Use
git fetch - all
git checkout v3.9.0
. /PF-build.sh

3d-gussner, 2020-08-13:
Lot of people use the PF-build.sh to compile their own firmware and it never let me/them down.

Hope I could prove that this script will get you as close as possible to what Prusa build server generates.

If you change the source code I would suggest that you also change the build number in Configuration.h before compiling.
The script will then generate new folders and names, which makes it easier to see which build works which not.

Also an awesome project to test and verify your changes is https://github.com/vintagepc/MK404
I use it now for months and helped a lot to see if the changes working correctly.
It cannot simulate everything but a lot.

3d-gussner, 2020-08-13:
In https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit you find the instructions for PF-build.sh 😉

ANTALIFE, 2020-08-14:
> Use
git fetch - all
git checkout v3.9.0
. /PF-build.sh

Alright, compiled using "v3.9.0" branch and still get differences in same areas as before:
[image]

Looks like this is as close as I can get it when compiling on Fedora 5.7.14-200 with ". /PF-build.sh", so will call this a win and continue ([jsw on Prusa forums](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3s-3-9-0-compiled-firmware-does-not-match-prusa-release/#post-245311) said differences are expected when compiling on different systems)

===================================================================================

> Also an awesome project to test and verify your changes is https://github.com/vintagepc/MK404
I use it now for months and helped a lot to see if the changes working correctly.
It cannot simulate everything but a lot.

That's a cool, a virtual MK3S. Thanks for showing me this, will definitely test on here before I upload to my printer 

===================================================================================

> In https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit you find the instructions for PF-build.sh 😉

Ah, but your guys [Linux instructions](https://github.com/prusa3d/Prusa-Firmware#linux) say to use "./build.sh" ;^)
[image]

closed 2020-08-14 by ANTALIFE