prusa3d/Prusa-Firmware #2801: After filament load, starts to extrude before returning to print

sledjunk, 2020-08-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

3d-gussner, 2020-08-12:
Thanks for the bug report but it is a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2502
Please consider next time to search all (also closed) issues if they may have been already addressed.

BTW Your Issue report is a top example how to fill out the template! Thanks for that.

sledjunk, 2020-08-12:
Sorry about the double post.  I guess I didn't read far enough.

JohnnyDeer, 2020-11-26:
Closing as duplicate of #2502.

closed 2020-11-26 by JohnnyDeer