prusa3d/Prusa-Firmware #2819: How to build for MK3S

nordurljosahvida, 2020-09-01:
I've checked out stable 3.9.1 on MK3 branch but my MK3S complains the firmware is for MK3:

`mk3 firmware detected on mk3s printer`

how do I build specifically for MK3S?

Thanks!

RobReus, 2020-09-02:
Its documented in the readme, see: https://github.com/prusa3d/Prusa-Firmware#2b

You have to copy the configuration for mk3s as the `build.sh` script looks for the existence of the file `Configuration_prusa.h` and if it does not exist, uses the config for the mk3 as default.

nordurljosahvida, 2020-09-04:
Thanks for this, i had assumed that since there was no specific branch for MK3S point 2b wasn't applicable to me. Sorry for the noise and thanks again.

closed 2020-09-04 by nordurljosahvida