prusa3d/Prusa-Firmware #2822: 3.9.1-RC1. Octoprint caused M112

funkyspider, 2020-09-03:
MK3S
3.9.1-RC1
Octoprint

Following installation of 3.9.1 RC1 I received an M112 when driving the printer from OctoPrint
Pi 4
Octoprint : 1.4.2
baud rate : 115200
1metre USB cable.

funkyspider, 2020-09-03:
Hi.  The issue happened whilst printing. Layer 2 of a relatively small model.
I will check if I have any logs.  I only ever saw this 112 once with 3.9.0.  I will swap USB cables.

funkyspider, 2020-09-03:
Log attached.  
Looks like it may have detected a thermal issue, so may have been a genuine stop.

| Send: N3911 G1 X56.912 Y86.995 E0.02369*84
| Recv: Error: HOTEND THERMAL RUNAWAY
2020-09-02 21:55:32,158 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: HOTEND THERMAL RUNAWAY"
2020-09-02 21:55:32,168 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2020-09-02 21:55:32,182 - octoprint.filemanager.analysis - INFO - Starting analysis of local:files-lager-22-2x_0.3mm_PLA_MK3S_3h28m.gcode
2020-09-02 21:55:32,238 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/files-lager-22-2x_0.3mm_PLA_MK3S_3h28m.gcode
2020-09-02 21:55:32,283 - octoprint.util.comm - INFO - Changing monitoring state from "Error: HOTEND THERMAL RUNAWAY" to "Offline (Error: HOTEND THERMAL RUNAWAY)"
2020-09-02 21:55:32,317 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-02 21:55:32,396 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2020-09-02 21:55:32,502 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_firmware, payload: {'commerror_text': u'HOTEND THERMAL RUNAWAY'}
2020-09-02 21:55:32,676 - octoprint.plugins.tracking - INFO - Sent tracking event print_failed, payload: {'origin': u'local', u'reason': u'error', u'commerror_text': u'HOTEND THERMAL RUNAWAY', 'file': '6fdd47879c8f2ee309cda48a5063b8e0413157e7', u'elapsed': 1142}

[octoprint.log.2020-09-02.log](https://github.com/prusa3d/Prusa-Firmware/files/5168690/octoprint.log.2020-09-02.log)

leptun, 2020-09-03:
@funkyspider Then it wasn't a communication issue in your case and you don't need to change the USB cable. Do you have a MK3S as well or is it a MK2.5S?

leptun, 2020-09-03:
Btw, this isn't exactly the log I need. What I need is a dump of the Serial exchange between Octoprint and the printer.
It's called `serial.log` in these instructions: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

hernandez87v, 2020-10-07:
Hey I have an MK2.5S with same issue and have details on #2490 but can move to here since its newer ticket.
Heres my serial.log, I set the heatbed temp only, to 50c:
[serial.log](https://github.com/prusa3d/Prusa-Firmware/files/5343212/serial.log)

From other ticket #2490 

I just updated my MK2.5s to latest. (as a side note I've been running MK2.5 FIRMWARE because I havent fully upgraded my extruder but it has neve been an issue)

- FIRMWARE_NAME: prusa3d_fw_MK25_3_9_1_3518_RAMBo13a_en-es.zip

neophyl, 2020-10-08:
That looks like mechanical strain on the conductors where they exit the printed clamp.  As the individual conductors break the remaining have to handle more current which means they get warmer.  Eventually when enough break they get very warm and cause that insulation damage.

When you redo the connections I'd guess that you will find similar though not as bad fatigue on the positive cable too.  Those wires need cutting back and new crimps attaching for sure.  Soldering is NOT recommended to reattach to connectors designed to be crimped onto wire.

JohnnyDeer, 2020-12-01:
@funkyspider is this issue still actual or can be closed? Thank you.

funkyspider, 2020-12-01:
I have been unable to reproduce the issue to date. Please close and I will
re-raise if I can reproduce it reliably.

closed 2020-12-02 by JohnnyDeer