prusa3d/Prusa-Firmware #2825: [BUG] Read-only SD card (write protected); print failure

sudo-gedit, 2020-09-07:
I have spent many many many hours to find a problem that does not exist!

problem:
Since the SD card in my printer was write-protected, the printer simply canceled the print without a message when the filament was changed or in between.

solution:
Please implement a check when starting the printer and also when inserting the SD card. If the SD card is then write-protected, an error message should appear.

My printer: Prusa I3 MK3S with MMU2S
Printer firmware: 3.9.0
MMU firmware: 1.0.6

The problem can be reproduced.

leptun, 2020-09-07:
Write protected in software (SD card's CSD register) or the Write protect physical switch?
Because I can assure you that the hardware can only detect if the card is inserted, but never if it is write protected in hardware or not. If the switch _really_  made a difference, then you might have a faulty LCD assembly.

sudo-gedit, 2020-09-07:
Sorry
I meant the hardware switch on the SD card.

Should I contact support so that the "LCD assembly" can be exchanged?

leptun, 2020-09-07:
Did you try to use a different SD card freshly formatted as FAT32?

sudo-gedit, 2020-09-07:
I can test that on Thursday at the earliest.

DRracer, 2020-09-21:
@sudo-gedit this is definitely a HW issue, either the SD card or the slot in the LCD. 
We tried a few SD cards in the read-only (locked) setup (lever down) and all of them worked fine and were sending data to the FW.
Closing this issue...

closed 2020-09-21 by DRracer