prusa3d/Prusa-Firmware #2826: [BUG]Prompt for Filament change 4 times.. no M600 in gcode

tommck, 2020-09-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0-3421

**MMU Upgrade** - No

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
I had a print prompt me to change the filament 3 or 4 times during a print. There were no M600 commands in the gcode.
  
**To Reproduce**
Print the gcode 
  
**Expected behavior**
I expected it not to prompt me for filament change when it wasn't supposed to happen
  
**G-code**
[gcode on Google Drive](https://drive.google.com/file/d/15RtL_yaBvfSEpPVhXkYzJ4TLmiK6_Hyp/view?usp=sharing)

**Video**
N/A

leptun, 2020-09-07:
There doesn't need to be a `M600` command for the filament switch routine to initialize. It is possible that there is a problem with your filament sensor where it usually senses the filament sensor, but fails randomly. If the firmware detects the lack of filament, it asks for replacement immediately.

tommck, 2020-09-07:
@leptun I would have expected it to have a different message when it detects the loss of filament versus an expected filament change (through M600).

3d-gussner, 2020-09-07:
Please check your IR sensor, cable and sensor lever.

ochm, 2020-09-14:
Turn off filament sensor and try to print. IMHO the unwanted filament change will not occur again.

DRracer, 2020-09-21:
@tommck Did turning off the filament sensor solve your issue? Did you check the wiring, lever etc.?

tommck, 2020-09-21:
Turning off the sensor let me finish the print. Turned it back on fire the
next print. Had no problems since...

DRracer, 2020-09-22:
@tommck ok, thank you for your report, closing this issue

closed 2020-09-22 by DRracer

tommck, 2020-09-22:
HA! @DRracer, it's like you jinxed me... It just happened again on another print... I disabled the filament sensor again... :(