prusa3d/Prusa-Firmware #2830: [ENHANCEMENT] Automatically raise extruder after the print is completed

Bill2462, 2020-09-11:
Printer : MK3S, 

Problem:
When the print is cancelled/finished, extruder parks very close to the bed surface. This can make removing the print surface a bit tricky and can lead to damage to the hotend wiring if user is to lazy to raise it using axis movement option.

Solution:
Usually I go to settings->move axis and raise the extruder to around 80mm above print surface before I remove it.
I think this can happen automatically when print finishes or is cancelled.

Bill2462, 2020-09-11:
Edit:
I just noticed that that happens only when print is aborted. When print finishes normally, extruder moves all the way to the left (y=0) and that is no longer a problem. Not sure whether this is a bug or not.

leptun, 2020-09-12:
Doesn’t the nozzle raise 10mm before parking? Or is that not enough?

Bill2462, 2020-09-12:
It raises by 10mm but this is not enough to remove print surface freely if print is not very tall or if it fails on one of the first few layers.

It seems that when the print finishes normally, extruder moves to all the way to the left and I can remove the surface just fine.
But when the print is aborted, extruder does not move all the way to the left and it stays above the build platform.
Yes it raises by 10mm but this is often not enough.

I will test this tomorrow and provide some pictures but it also seems that there is a difference in y axis movement.

leptun, 2020-09-12:
The parking position defined in the gcode file for when the print finishes is not part of the firmware, but rather just 2 move instructions.
In firmware it is set up differently than it is in gcode.

Btw, there is a quick shortcut fo raising the nozzle more. Just press and hold the knob for 1s and the move Z menu will open

wavexx, 2020-09-22:
I agree with @Bill2462 here. I modified the end_gcode in the slicer and the stop code in the FW to raise at least 5cm above the current print position so that the steel sheet can be removed easily.

Changing the end_gcode is easy. On the FW I'm using this:

https://github.com/prusa3d/Prusa-Firmware/commit/3a3e65e057e0284ebd30072f180ba3bae67ca979

but for a PR maybe we should define the LIFT amount in the variant file instead of adding yet another magic value..

closed 2023-11-01 by github-actions[bot]