prusa3d/Prusa-Firmware #2831: [BUG] long sustained beep on inital warmup (regression)

iplayfast, 2020-09-13:
MK3 MMU2s 
version 3.9
USB/Octoprint
When starting a print from cold state, there is a long sustained beep until the extruder reaches about 170. 

I backed off to firmware 3.8, and did not notice it there. (although that might be chance). 

**To Reproduce**
Start a print using pla

  
**Video**
  Please attach a video. It usually helps to solve the problem.

iplayfast, 2020-09-13:
It's possible, that I may have uploaded mk3 firmware instead of the mmu2s version. 
Reloaded mmu2s version, and problem is gone. This I believe this is still a bug, as there is no indication of why it's beeping. (wrong firmware, unknown hardware detected? I don't know).

leptun, 2020-09-14:
Can you please make a video of the issue?

3d-gussner, 2020-09-14:
@iplayfast Can you check which printer and mmu2 firmware you have flashed?
- Open Pronterface or Octoprint and connect to your printer
- You should see in the terminal output something like
```
MMU => 'start'
MMU <= 'S1'
MMU => '106ok'
MMU <= 'S2'
MMU => '372ok'
MMU version valid
MMU <= 'M1'
MMU - ENABLED
```
where `106` means that the mmu2 has firmware 1.0.6. installed
- send g-code `M115` to the printer and you should see something like
```
SENDING:M115
FIRMWARE_NAME:Prusa-Firmware 3.9.1-RC1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
```

3d-gussner, 2020-09-22:
@iplayfast Any update on this issue?

iplayfast, 2020-09-25:
Sorry I missed the notification of your request. 

video https://www.dropbox.com/s/3u1gxdwnn7tvqa1/20200905_191748.mp4?dl=0

Recv: FIRMWARE_NAME:Prusa-Firmware 3.9.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
....
Send: M105
Recv: ok T:25.5 /0.0 B:26.0 /0.0 T0:25.5 /0.0 @:0 B@:0 P:27.8 A:27.1
Recv: MMU => 'start'
Recv: MMU <= 'S1'
Send: M105
Recv: ok T:25.6 /0.0 B:26.0 /0.0 T0:25.6 /0.0 @:0 B@:0 P:26.5 A:27.3
Recv: MMU => '106ok'
Recv: MMU <= 'S2'
Recv: MMU => '372ok'
Recv: MMU version valid
Recv: MMU - ENABLED
Send: M105

DRracer, 2020-09-25:
@iplayfast is the beep really coming from the LCD beeper? This looks more like a PWM interference of some kind.

iplayfast, 2020-09-25:
I just tried again, with the lcd screen unplugged. Still beeping. So that leaves steppers, or ?.

I can't tell where it's coming from. For a while I suspected the power supply had some over load warning on it. It only happens when it's warming up. This is a new problem since "upgrading" to the mmu which takes a bit more power.
I also tried downgrading the firmware, and the problem seemed to go away, but came back again.

DRracer, 2020-09-25:
@iplayfast ok, so we can exclude the beeper. Also, with high probability, we can exclude FW as well, since you tried different versions and the problem didn't go away.
Can you put you ear beside the PSU and listen if the sound comes from there?
Also, try to disconnect the MMU - if the beeping still occurs with the MMU OFF

iplayfast, 2020-09-25:
I've tried disconnecting the MMU while the screech was happening. It continued, so it's not from the mmu or the steppers on that. I've tried putting my ear beside the PSU, the sound is very high and I can't figure out where it's coming from.  
I downloaded a sound meter for my phone, and of course, it's stopped doing it. (for now). Seems intermittent.  I'll let you know if I discover anything.

leptun, 2020-09-26:
Does the printer go through a UPS or some kind of smart power switch?

iplayfast, 2020-09-29:
as a matter of fact it does!

leptun, 2020-09-29:
I think the beeping is comming from the UPS then. Could it be entering some kind of power overload error? Try bypassing it and see if it improves

DRracer, 2020-10-01:
@iplayfast It looks like the beep is not caused by the printer/MMU itself, would you mind closing this issue?

closed 2020-11-03 by iplayfast