prusa3d/Prusa-Firmware #2834: [BUG] M601 - Head moves away from Object as expected and after a short time back to last Position

wseifert, 2020-09-16:
Entering a M601 at specific layer in Prusaslicer causes the printer (MK3S + MMU2, FW 4.9 & FW 4.9.1 RC1) to move the head to left back position to clear the area over the print object but after ~2 secs the head is moved back to position where the print stopped. Inserting a nut for example is impossible because the head blocks the area.
In gcode only M601 is visible:

;AFTER_LAYER_CHANGE
;46.7
M117 Mutter einsetzen und Druck wieder aufnehmen
M601
G1 Z46.900
G1 X232.256 Y61.382
G1 Z46.700
G1 E3.00000 F3600.00000
M204 S800
G1 F3000.000
G1 X232.256 Y60.843 E0.02925
G1 X232.545 Y61.196 E0.02479

leptun, 2020-09-16:
Are you printing from SD or from USB?
Also, please post the whole gcode file

wseifert, 2020-09-16:
I print via Octoprint (serial connection via GPIO).
[Plexi Box upper Part 2 mm_0.3mm_PETG_MK3SMMU2S_6h2m.zip](https://github.com/prusa3d/Prusa-Firmware/files/5230972/Plexi.Box.upper.Part.2.mm_0.3mm_PETG_MK3SMMU2S_6h2m.zip)

leptun, 2020-09-17:
I tried to reproduce the issue but couldn't. It did as I would expect on both SD printing and USB printing (octoprint). The fact that it was printed from a pi Zero from the rPI serial port is irrelevant in this case.

wseifert, 2020-09-17:
What could be the reason for this behaviour on my MK3S? He is running FW 3.9, no Mods.
So I will try to record the terminal output in Octoprint to take a look at the G codes are reported there.

wseifert, 2020-09-17:
I posted this issue also in Prusa Forum, a response was posted stating that other(s) also have this problem: [https://forum.prusaprinters.org/forum/original-prusa-i3-mk3-hilfe-zur-hardware-firmware-und-software/druckpause-mit-m601-druckkopf-faehrt-weg-und-faelschlicherweise-nach-kurzer-pause-wieder-zurueck-zum-druckobjekt/#post-264235](url)

wseifert, 2020-09-18:
I printed with serial.log enabled and found following:

.
.
.
2020-09-17 18:20:29,730 - Send: N187599 M117 Plane in Nuten einsetzen und Druck wieder aufnehmen*48
2020-09-17 18:20:29,739 - Recv: ok
2020-09-17 18:20:29,741 - Send: N187600 M601*28
2020-09-17 18:20:30,720 - Recv: echo:enqueing "G1 E-1.000 F2700"
2020-09-17 18:20:30,747 - Recv: // action:paused
2020-09-17 18:20:30,748 - Printer signalled that it paused, switching state...
2020-09-17 18:20:30,749 - Changing monitoring state from "Printing" to "Pausing"
2020-09-17 18:20:36,769 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2020-09-17 18:20:36,781 - Send: N187601 G1 Z46.900*126
2020-09-17 18:20:36,788 - Recv: ok
2020-09-17 18:20:36,792 - Send: N187602 G1 X232.446 Y150.613*22
2020-09-17 18:20:36,819 - Recv: ok
2020-09-17 18:20:36,821 - Changing monitoring state from "Pausing" to "Paused"
2020-09-17 18:20:36,828 - Send: M105
2020-09-17 18:20:36,835 - Recv: ok T:220.4 /0.0 B:89.9 /90.0 T0:220.4 /0.0 @:0 B@:61 P:36.5 A:46.6
2020-09-17 18:20:36,840 - Send: M105
2020-09-17 18:20:36,869 - Recv: ok T:220.3 /0.0 B:89.8 /90.0 T0:220.3 /0.0 @:0 B@:81 P:36.5 A:46.8
2020-09-17 18:20:38,562 - Send: M105
2020-09-17 18:20:38,570 - Recv: ok T:219.6 /0.0 B:90.0 /90.0 T0:219.6 /0.0 @:0 B@:51 P:36.5 A:46.7
2020-09-17 18:20:43,565 - Send: M105
2020-09-17 18:20:43,573 - Recv: ok T:218.8 /0.0 B:90.0 /90.0 T0:218.8 /0.0 @:0 B@:43 P:36.4 A:46.6
2020-09-17 18:20:48,567 - Send: M105
2020-09-17 18:20:48,575 - Recv: ok T:216.7 /0.0 B:90.1 /90.0 T0:216.7 /0.0 @:0 B@:35 P:36.4 A:46.5
2020-09-17 18:20:53,571 - Send: M105
2020-09-17 18:20:53,580 - Recv: ok T:214.0 /0.0 B:90.1 /90.0 T0:214.0 /0.0 @:0 B@:52 P:36.4 A:46.6
2020-09-17 18:20:58,575 - Send: M105
2020-09-17 18:20:58,588 - Recv: ok T:210.4 /0.0 B:90.0 /90.0 T0:210.4 /0.0 @:0 B@:58 P:36.5 A:46.6
2020-09-17 18:21:03,578 - Send: M105
2020-09-17 18:21:03,586 - Recv: ok T:207.3 /0.0 B:89.9 /90.0 T0:207.3 /0.0 @:0 B@:61 P:36.5 A:46.5
2020-09-17 18:21:08,580 - Send: M105
2020-09-17 18:21:08,589 - Recv: ok T:204.1 /0.0 B:90.0 /90.0 T0:204.1 /0.0 @:0 B@:53 P:36.5 A:46.5
2020-09-17 18:21:13,582 - Send: M105
2020-09-17 18:21:14,365 - Recv: T:200.0 E:0 W:2
2020-09-17 18:21:15,367 - Recv: T:198.7 E:0 W:2
2020-09-17 18:21:16,369 - Recv: T:197.1 E:0 W:2
2020-09-17 18:21:17,371 - Recv: T:195.7 E:0 W:2
2020-09-17 18:21:18,373 - Recv: T:194.8 E:0 W:2
2020-09-17 18:21:19,376 - Recv: T:194.1 E:0 W:2
2020-09-17 18:21:20,377 - Recv: T:193.9 E:0 W:2
2020-09-17 18:21:21,379 - Recv: T:194.1 E:0 W:2
2020-09-17 18:21:22,381 - Recv: T:194.5 E:0 W:2
2020-09-17 18:21:23,383 - Recv: T:195.0 E:0 W:2
2020-09-17 18:21:24,386 - Recv: T:196.1 E:0 W:2
2020-09-17 18:21:25,389 - Recv: T:196.9 E:0 W:2
2020-09-17 18:21:26,391 - Recv: T:197.6 E:0 W:2
2020-09-17 18:21:27,394 - Recv: T:198.6 E:0 W:2
2020-09-17 18:21:28,396 - Recv: T:199.7 E:0 W:2
2020-09-17 18:21:29,399 - Recv: T:200.9 E:0 W:2
2020-09-17 18:21:30,401 - Recv: T:202.2 E:0 W:2
2020-09-17 18:21:31,421 - Recv: T:203.4 E:0 W:2
2020-09-17 18:21:32,405 - Recv: T:204.6 E:0 W:2
2020-09-17 18:21:33,416 - Recv: T:206.0 E:0 W:2
2020-09-17 18:21:34,410 - Recv: T:207.2 E:0 W:2
2020-09-17 18:21:35,411 - Recv: T:208.8 E:0 W:2
2020-09-17 18:21:36,413 - Recv: T:209.2 E:0 W:2
2020-09-17 18:21:37,418 - Recv: T:210.1 E:0 W:2
2020-09-17 18:21:38,466 - Recv: T:211.0 E:0 W:2
2020-09-17 18:21:39,469 - Recv: T:211.7 E:0 W:2
2020-09-17 18:21:40,469 - Recv: T:212.8 E:0 W:2
2020-09-17 18:21:41,470 - Recv: T:213.4 E:0 W:2
2020-09-17 18:21:42,473 - Recv: T:213.4 E:0 W:2
2020-09-17 18:21:43,484 - Recv: T:213.2 E:0 W:2
2020-09-17 18:21:44,477 - Recv: T:213.1 E:0 W:2
2020-09-17 18:21:45,478 - Recv: T:213.3 E:0 W:2
2020-09-17 18:21:46,481 - Recv: T:213.9 E:0 W:2
2020-09-17 18:21:47,482 - Recv: T:213.8 E:0 W:2
2020-09-17 18:21:48,485 - Recv: T:213.9 E:0 W:2
2020-09-17 18:21:49,486 - Recv: T:214.3 E:0 W:2
2020-09-17 18:21:50,489 - Recv: T:214.4 E:0 W:1
2020-09-17 18:21:51,490 - Recv: T:214.9 E:0 W:0
2020-09-17 18:21:56,327 - Recv: Resend: 187601
2020-09-17 18:21:56,334 - Recv: ok
2020-09-17 18:21:56,336 - Send: N187601 G1 Z46.900*126
2020-09-17 18:21:56,336 - Recv: ok
2020-09-17 18:21:56,339 - Send: N187602 G1 X232.446 Y150.613*22
2020-09-17 18:21:56,340 - Recv: // action:resumed
2020-09-17 18:21:56,341 - Printer signalled that it resumed, switching state...
2020-09-17 18:21:56,341 - Changing monitoring state from "Paused" to "Resuming"
2020-09-17 18:21:56,354 - Recv: ok
2020-09-17 18:21:56,355 - Changing monitoring state from "Resuming" to "Printing"
2020-09-17 18:21:56,362 - Send: N187603 G1 Z46.700*114
2020-09-17 18:21:56,380 - Recv: ok
2020-09-17 18:21:56,390 - Recv: ok
2020-09-17 18:21:56,392 - Send: N187604 G1 E3.00000 F3600.00000*33
2020-09-17 18:21:56,404 - Send: N187605 M204 S800*83
2020-09-17 18:21:56,405 - Recv: ok
2020-09-17 18:21:56,408 - Recv: ok
2020-09-17 18:21:56,411 - Send: N187606 G1 F1800.000*103
2020-09-17 18:21:56,424 - Recv: ok
2020-09-17 18:21:56,426 - Send: N187607 G1 X232.636 Y150.261 E0.02672*93
2020-09-17 18:21:56,437 - Recv: ok 

It looks like that Octoprint claims Communication timeout while printing and does not correctly interpret the messages received from printer and sends travel G-codes which causes the wrong head position.

Do you know how to solve this or have I to go to Octoprint forum?

wseifert, 2020-09-18:
I send you the complete serial.log
[serial.log.zip](https://github.com/prusa3d/Prusa-Firmware/files/5245002/serial.log.zip)

leptun, 2020-09-18:
Hmmm... I’ve encountered this issue before. It was related to pause again, but this time during the fanCheck error. I’ll have to think of a way to let octoprint know quicker that it needs to dump it’s queue.
What octoprint version do you have? Maybe I can generate the issue there

wseifert, 2020-09-18:
Octoprint 1.4.2

sarf2k4, 2020-10-20:
I am also experiencing this issues, I am using latest mk3s and mmu2s firmware with octoprint, octoprint is connected to the printer via usb. I can confirm that any print job with m601 from octoprint, will have the nozzle goes back to the print location (or the next gcode command). Printing directly from sd are okay.

It is as if that the einsy firmware just executed the gcodes in the buffer sent via usb, like 2 lines of gcodes after M601 has been issued or read, and the firmware kind of "too late" to catch M601 that is pause before the next gcode after M601.

wseifert, 2023-10-25:
issue solved in following versions, no longer present

closed 2023-10-25 by wseifert