prusa3d/Prusa-Firmware #2839: [BUG] Crash detection not working with firmware 3.9.1 (MK3S)

NoNme1, 2020-09-22:
Hello,
Prusa support directed me here to submit a bug report. 
Printer: MK3S
FW version: 3.9.1 (latest release from the website)
No MMU
printed via SD card 

Crash detection on MK3S not working after I uploaded the new firmware. First noticed the issue on a tpu print that curled up and caused the toolhead to crash on the print and the x-axis to skip steps (didn't test on the Y-axis). the printer is on Normal mode and Crash det. is [on]
[crash detection not working.zip…]()
included in the zip is a video of the issue and the gcode used
crash detection does work when I flash 3.9.0 and print the same gcode.

Steps to reproduce:
preheated FLEX > Print > and selected the gcode, once the print has started, moving the toolhead does not cause the printer to re-home and resume the print as expected.

MohanMaker, 2020-09-22:
I have the same issue here with firmware 3.9.1 on Prusa MK3S. Thanks for submitting the issue report @NoNme1

SteWhite, 2020-09-29:
me too (3.9.1 on Prusa MK3S)

nikil511, 2020-12-19:
Same here, with 3.9.2
(was working with older firmware)