prusa3d/Prusa-Firmware #2843: [BUG] 3.9.1 - LA 1.5 Saga continues...?

vintagepc, 2020-09-23:
**Printer type** - MK3S
**Printer firmware version**- 3.9.1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint**
SD

**Describe the bug**
While we've made leaps and bounds over the previous versions, I just did a print that appears to still suffer from LA1.5 issues with small extrusions/starts/stops.

I did run the  LA calibration g-code and obtained a K value of 0.08, which agrees with the release notes as a valid default.
However, the lettering on the print suffers from massive overextrusion: 
[image]
  
For reference, here are some previous PETG filament chips I made with previous firmware versions. 

[image]

Pronterface snap of the region:
[image]

**To Reproduce**
[Prusa_Red_PETG-0.2 PETG 24m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/5269823/Prusa_Red_PETG-0.2.PETG.24m.gcode.zip)

I printed the example G-code with Prusament Carmine Red, at 235C, K 0.08

Stock Prusament profile, except the K value was modified, and the temperature dialled back to (unsuccessfully) counter stringiness). 
  
**Expected behavior**
Non-overextruded lettering

Kachidoki2807, 2020-09-24:
Can you provide the .3mf of this project?
I guess this is an issue more PrusaSlicer related (2.3.0-alpha0) or special settings, the printer has very weird travel moves, like an avoid crossing perimeters option enabled.

Below my print of your G-Code, not the best print I made on this printer, but the quality is quite decent. Prusament PETG Galaxy Black.
[image]

Yes, maybe there is a bit overextrusion, but it can also be the small cumulated oozing during those weird travel move that take ages.

vintagepc, 2020-09-24:
Yes, I do have 'avoid crossing perimeters' ticked; I know the travel paths are less optimized. 

Yours turned out better than mine, though the letters are also far stringier than expected. 

Standby; I resliced and am testing something now; will provide 3MF if that also has issues, or follow up if it resolves it.

vintagepc, 2020-09-24:
Sorry for the false alarm. Turns out it was my fault; at some point through errant keypresses or some other means, the 'Extra distance after retraction' value had been changed to 0.2; Of course this caused overextrusion on de-retract. 

I changed that back to 0 and the print quality is acceptable again for PETG with the usual slight stringiness.

closed 2020-09-24 by vintagepc