prusa3d/Prusa-Firmware #2844: [FEATURE REQUEST] Advanced G80/G81 for bed leveling

pedjas, 2020-09-28:
**Is your feature request related to a problem? Please describe.**

When doing bed leveling with nylocks (see: https://github.com/PrusaOwners/prusaowners/wiki/Bed_Leveling_without_Wave_Springs) updating levels of 7x7 points on bed takes quite a time and user has to deal with all points at once, or if he needs to adjust just one point he has to wait for measurements to be done on all points.
  
**Describe the solution you'd like**

Some other manufacturers offer bit simplified and quicker method to read levels: reading single, specified, point. 

If user can run measuring level of single point he could do adjustment point by point which should be much quicker. He would repeat measurement of one point unitl he gets it right and then move to next point and next and next...

wavexx, 2020-10-05:
You can use G30 for this: https://reprap.org/wiki/G-code#G30:_Single_Z-Probe

pedjas, 2020-10-13:
Hm, that is feasible. Where can I find coordinates of measurement points used for G80/G81 measurements?

marcolivierroy, 2020-10-13:
I use this since I could not get G30 working all the time for an unknown reason and G81 is not enabled in the firmware. Hope it helps.
G28 W ; home all without mesh bed level
G80     ; mesh bed leveling

closed 2023-11-01 by github-actions[bot]