prusa3d/Prusa-Firmware #2846: [BUG] MMU2S filament load and change fails 90% of the time

linzwatt, 2020-09-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.8.1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**SD card or USB/Octoprint** - SD CARD

**Describe the bug**
When starting a print, the first filament load fails most of the time, if the first filament load works, any filament changes during the print are most likely to fail too. The MMU2S pushes the filament all the way into the extruder, activates the sensor (confirmed by looking at sensor debug menu), filament goes back and forth a little then is drawn all the way back into the MMU2S. This repeats 3 times before a filament load failure is reported on the LCD. 

However, when I do a "Load to nozzle" before starting a single filament print, this works 100% of the time. 
If you start a print straight after a "Load to nozzle", selecting the same filament you loaded to nozzle, the print starts straight away without loading again.

I imagine the "Load to nozzle" uses a slightly different procedure to the loading process at start of print and filament change process during print. Can any developers confirm whether the process is different? Or perhaps the timing is different? It could be that the sampling window for extruder filament sensor is not long enough during a load at start of print.
  
**To Reproduce**
1. Start a print with no filament loaded
2. Printer will try to load the selected filament - likely to fail
3. Stop the print
4. Do a "Load to nozzle" command
5. Printer will load the selected filament - likely to work 1st time
6. Start a print
7. Printer starts the print with the pre-loaded filament, no issues
8. Subsequent filament changes during the print are still likely to fail
  
**Expected behavior**
Printer should load filament successfully 100% of the time

sudo-gedit, 2020-09-29:
I can confirm the mistake too.

dikodahan, 2020-09-30:
Not sure if this is tied to the same issue listed above, but I am experiencing a similar issue at the end of a print.
This has just started after the upgrade to MK3S 3.9.1 (MMU2S at 1.0.6). The entire print will complete successfully and then in the unload at the end I am getting an error message saying MMU2 needs attention and I have to manually pull the filament out.

The bigger issue is that the MMU is "stuck" after. The only way to get it back to normal operation is a complete disconnect and reconnect from power. A simple button reset does not work!

dikodahan, 2020-10-02:
Any update on this one??
This is a seriously irritating behavior!

Syl20-94, 2020-10-06:
Hello, I've got tha same bahaviour, with FW 3.9.1 and MMU 1.0.6, each action on the MMU cause strange behave and all leds of mmu unit blink fast (orange+green) then MMU reboot a couple of time before going  back on the normal way - usually I need to reset manually the MMU to go on.

dikodahan, 2020-10-06:
I can't do anything now with the MMU... any type of action causes all LEDs to blink orange.
Prusa team, help here please...??

linzwatt, 2020-10-07:
I think the most recent comments are separate issues, although they all relate to very unreliable MMU firmware
would be good to get some comments from the main contributors to the MMU firmware

@wavexx @josefprusa @PavelSindler @DRracer @mkbel ?

ANTALIFE, 2020-10-11:
Sounds like you have bad tips, bad as in the diameter is irregular. Not long ago I made some changes to 3.9.0 to get around this

**https://www.antalife.com/2020/08/project-mmu2s-filament-diameter.html**

sudo-gedit, 2020-10-16:
Is there no interest in fixing the problem?

Syl20-94, 2020-10-16:
@ANTALIFE , your point is interesting, but I'm not sure about it. in the process, filament is loading and detected by both PINDA and FINDA sensor (as motor MMU stop loading)  then it goes back and do a reload twice with the same process (is it a kind of loop ??) and goes in failure (all led blinking) which need to reset the MMU unit. 
after the reset and without touching the filament (neither cutting/replacing) the MMU is loading normally (but sometime I need to reset a couple of time again) 

So Yes I'm interested in fixing this problem but I'm not the developper so I can not goes deeper in fixing the only thing I can do is living with this workaround which is irritating me on each printing

ANTALIFE, 2020-10-16:
@Syl20-94 Show us how your tips look when you see a load error/loop

mkbel, 2020-10-16:
@sudo-gedit Can you supply log of the problem? How to obtain log: https://github.com/prusa3d/Prusa-Firmware/issues/1689#issuecomment-680850966

dikodahan, 2020-10-16:
Guys, if it was a bad tips issue then the MMU2 would have had issues prior to 3.9.1 firmware. I've had the MMU2 (and then upgraded to MMU2S) for 2 years now. no issue at all, up until I upgraded to 3.9.1 on the printer, all of a sudden my tips are the issue??

In addition to the above, I have also, as an additional precautionary measure, took all the filaments out from it, clipped them, and put them back in. still the same problem.

mkbel, 2020-10-16:
@dikodahan Can you supply log of the problem? How to obtain log: https://github.com/prusa3d/Prusa-Firmware/issues/1689#issuecomment-680850966

Syl20-94, 2020-10-17:
@ANTALIFE indeed I can say it's a tip, here is my workflow.
1- Prepare print through PrusaSlicer 2.2.0, my printer is a mk3S with FW 3.9.1
2- Send Gcode to octoprint 0.17 from Prusaslicer via network
3- On octoprint, launch the print and select filament  via MMU2S Select Filament plugin
From this point printer and octoprint work in automatic
4- Octoprint set heating for extruder and bed, in the meantime, 
5- Filament is loading, MMU unit goes to a start-up sequence and move from left to right and move to correct filament position, start to load the filament to the extruder.
6- When filament arrive to the extruder - the lead screw is moving as filament is loaded by the extruder motor- MMU motor stop 
7- After a short time (1 or 2 sec) filament is unloaded from the extruder back to MMU unit
8- Then it redo step 5 to 7 two time
9- after 3 loops, filament is unload and MMU unit start to blink for all orange + green LEDs
10- I catch a screw pin and press reset button on the MMU unit 
11- MMU is rebooting and redo from step 5.
then it can redo either load/unload sequence til I need to reset MMU unit or it load correctly the filament 
when filament is loaded, print sequence is going on without problem (currently, I print in 1 color mode so without filament change during printing).

so I dont touch filament during the whole process.
feel free to tell me if you need logs and how to catch it

iplayfast, 2020-10-22:
I thought I was going mad. I'm glad to see that this is a bug that is affecting other people too.

mkbel, 2020-10-22:
@iplayfast I would be really happy to fix this problem. It would by **very** helpful if someone who is able to reproduce the problem would do so when having printer connected via USB cable to some computer and logging debug information present on that USB CDC communication channel. How to do that is described here: https://github.com/prusa3d/Prusa-Firmware/issues/1689#issuecomment-680850966

Syl20-94, 2020-10-24:
update from my side, I find out that the spring on the idler-body which is on the motor side was missing, I put it back and after 4-5 print, loading from scratch seems to be OK and it's goes back to normal working. I'm not explaining the reason but I recommand to check if the idler-body is not too tightened screwed (or too loosed)

linzwatt, 2020-10-25:
@mkbel I will supply some logging info in the next couple of days, thanks for your interest and the info about how to collect logs :)

chriswal, 2020-11-01:
if all leds are blinking on the MMU2 there is a problem with the motor drivers. if the 5v drops the drivers lose their configuration. this problem often comes from a loose plug. reseat your 5v plug on the mmu2 and the einsy board. or the drivers are overheated. touch your board of the mmu2 with your finger (but be carefull it could be very hot) if you burn it ;-) you know its overheated. (mine got 80degrees celsius in idle).

sudo-gedit, 2020-11-16:
are there any new findings?

mkbel, 2020-11-17:
@sudo-gedit Still no one produced log. I am main maintainer of MMUv2 firmware. I would be very happy to fix this bug if present. But I desperately need log to fix the problem.

dikodahan, 2020-11-17:
I will produce an output log by the weekend for you on this one.

lukasradek, 2021-03-27:
I have had the exact same issue and I have seemed to resolve it.
It is partially user error, but it seems to me, firmaware could do better to prevent such issues.

There are 3 things I have checked and readjusted.

1) IR sensor - make sure it can be triggered even by a 1.5mm allen key. Make it very sensitive. If you see, that during normal operation, the IR value is jumping between 0 and 1 (with filament still in extruder) - it causes problems.

2) FINDA probe - kind of the opposite. Make sure it can be trigger ONLY by full filament. Not by a string hanging on the end. This could/should be solved by adding more cooling moves and does not directly cause the above mentioned issue, but it was causing related problems, when filament unloaded too far (past hobbed pulley) and could not load again. 

3) MMU Idler drum position - my idler drum shaft aretation screws came a bit loose and there was about 15° play on the drum rotation. That meant that sometime the bearing did not put enough pressure on the pulley to push the filament, but most importantly when the bearing was suppost to move away (to hand of the filament control to the extruder), the bearing still applied some pressure, making it impossible for extruder to pull the filament (clicking noises).

So first check if the IR sensor does not give intermitent values while filament is loaded and than WHILE filament is loaded through the MMU, pull the filament and watch if it spins the bearing, which should be disengaged. If it does, you drum is probably loose.

Now for the firmware part.
It feels to me, that (for example while unloading filament from nozzle) the firmware waits for the IR value to be 0, then switches to "filament out of extruder" state and does not care about the value anymore. So if the value just jumped from 1 to 0 and back again to 1 due to not perfect IR calibration. The firmware assumes, the filament is out of extruder, because it saw 0 for a moment. But does not care that the value is back to 1 and proceeds with MMU retracting the filament, which strains the system and grinds the filament.

It feels to me like it is implemented as kind of a "one-way" state machine, that does not account for the possibility of inconsistent sensor value.

I have had similar thing happen to FINDA, when MMU kept giving me unload error (fast red blink), while the filament was already out of the selector. That happend to me only once though (as opposed to IR), so I cannot guarantee I have not overlooked something (possibly a string tripping the FINDA).

MathieuGuennaoui, 2021-04-07:
I'm currently facing the same problem (MK3S/MMU2S). 
Using the "Load the nozzle" trick is a workaround for single material prints, but multimaterial parts tend to fail because of that weird interaction between the MMU and the extruder. OP explained all the behavior quite accurately, and I double checked both the IR sensor positioning and the filament path cleanness. I would add that even the "Load the nozzle" option may fail for me. I just don't get why the extruder sends the filament back to the MMU, it could just print nicely because everything is actually fine. 
I tried with different filament, seems to me that it happens more often with the PETG galaxy black Prusament, especially compared to the orange PETG Prusament which usually doesn't fail, but I can't be truly reliable on that point. Did you get the same results with every filament?

chriswal, 2021-04-22:
i have not confirmed this by looking at the firmware, but in my opinion this is a problem between the tip iof the filament and the heatbreak. do you have the stock heatbreak or the mmu2s heatbreak? the problem with sending back the filament to the mmu2 is that on loading the filament the firmware pushes it way down the heatbreak. if there is a stall detected (the tip doesnt fit through the heatbreak) it tries to load again. if you switch on the cut function in the firmware, the firmware cuts the filament after the 3rd try, and possibly the tip will fit through.
the cause:
if i am not mistaken the drilling in the normal heatbreak and the tip is 2mm also the hole in the ptfe liner is 2mm.
if you unload the filament the filament is pressed into the tip and the cooled down to prevent strings on the tip of the filament.
now the filament tip has 2mm . when it further cools down it shrinks in one direction and expands in the other so the filament gets to 2,1 mm . now on the next load the filament can be pushed through the liner because its elastic, but it get stuck on the heatbreak.
the solution to this: prusa has already taken care of it with the mmu2s heatbreak.
i dont know the real measurements ..but the heatbreak should have a hole with a 2,1mm diameter and the ptfe liner a smaller 1,9mm. so if you create the tip of the filament on unload it should be about 2mm and fit through the heatbreak.

if the ir sensor doesnt work and the printer cant detect the "filament loaded" this can also happen.
i dont know if when you get the ir sensor signal and you loose it after detection what happens on the load.
you could check the reliability of the setting when you run the printer without the mmu and set the filament sensor to pause the print on filament runout. if it pauses randomly you have to adjust your sensor.

DaReaLDeviL, 2021-05-04:
Same problems here with the MMU2s. After print is finished 99% it need atention and the 1. filament loding problem. Also in the same boat like deiscirbed before.

snympi, 2021-05-29:
Similar problem encountered. Led installed on IR filament sensor so relatively easy to see what happens. MMU2S selector (mod with magnet) works great when loading and FINDA signals correctly when filament is pushed to extruder.

Extruder IR detects filament arrival, can see idler door move as well, then it seems to push filament forward and back, sometimes I can hear the bondtech gears grinding and the filament is sent back to MMU.

When printer was just MK3S+ loading filament by hand was always successful when then bondtech gears caught the filament. It would be great if MMU only loads filament to extruder and IR signals that it has filament. The additional tests when filament is pushed to hotend  seems to make my setup less reliable. The algorithm that signals successful load might be too complicated in some cases by detecting a false negative much more often than a false positive.

snympi, 2021-05-29:
@mkbel I'm late to this forum.just got my MMU2S and encounter the issues described above.You're the right person to talk to about understanding why the filament is pushed back and forth when the IR sensor triggers,instead of assuming it is loaded and continue to print.

What is needed to help you solve this?

mudkicker2, 2021-06-23:
@mkbel I have attempted your suggestion about logging.  I installed PUTTY, but when I click on "OPEN" my PC dings and nothing happens.  Is there another step I am missing??

DaReaLDeviL, 2021-06-24:
Printed the idler door and tried several times to resetup things. My only problem with the mmu2s is the idler door and its moving when loading the filament.

I can see it wooble back and forth a little bit and that is enough to get a "bad loading" and the filament is pushed out again. Please make it more reliable when it first sees the filament.  I struggle with the mmu2s now for over a year with that idler door. The screw for the idler door only has a small windows where it works and will come lose after some prints.

mudkicker2, 2021-06-24:
So I have the MKS3+ w/MMU2S. Mine was doing the following:
Load the filament from the mmu to the extruder. Once it got there, it would unload back to the mmu. It would repeat this cycle until it called up an error on the LCD.
What I found was this:
The long piece that is attached to the door has a perpendicular extended piece that "makes" the filament sensor. By watching the filament sensor status on the LCD, when I stuck a piece filament in there, it stayed at 0, instead of 1.
What I did to fix it:
I loosened the sensor and kept filament in the chute. Kept moving the sensor around until it maintained a 1. I then tightened the screw with the sensor in that position.
Conclusion:
Horrible design. I spent hours trying to figure this out. What happens is the motor keeps trying to pull filament through and it causes a clog in the nozzle. There needs to be a better system for this part.

snympi, 2021-07-24:
@mkbel any feedback on some of the recent questions above. Would like to help make the filament load in the extruder more reliable because current algorithm makes process unreliable.

gudnimg, 2021-07-24:
I don't have a MMU2S but I would love to help. It would be really nice to have the output from the Serial stream posted here. The firmware should print out what it is doing and when it sends the MMU commands. It is not quite clear to me where exactly in the state machine the issue is at.

Has anyone tried Putty or TeraTerm, or something similar to see what it is outputting?

mudkicker2, 2021-07-27:
@gudnimg I tried to use PUTTY, as I said in an earlier reply.  It doesn't work for me.  Also, I have a new issue that is matching what some of the previous people have stated.  When the printer is either changing filament or ending the print, the MMU is errored and "needs attention".  I found that the issue is that the filament end has formed a bulge on the tip and causes too much resistance for the MMU to pull it through the tube.  Could this be an issue with the nozzle, or is the filament sitting in the nozzle too long and allowed to cool before it gets pulled?

Ic33s, 2022-06-29:
After an other session (about an hour) to get any filament loaded again. I want to step in and want to ask if someone of Prusa cares about MMU2S+ anymore?!? The behaviour of the logic, when the filament triggers the IR sensor is way to sensitiv to sensor errors I would say. I want to screem on my printer he just shove the filament 1mm or 2mm more into the gears and everything is fine. Until know the MMU2S(+) was a real pain and takes the fun away from the whole mk3s+.

An yes everything is assembled correctly. Everything is aligned. The filament is moving without any problems to the feeder gears. It's no hardware issue. It's just a bad logic during loading the filament to the nozzle at the exact moment, when the filament reaches the gears. I even calibrated the fast filament length for all 5 filaments by using the MMU2 instruction.

Sorry, but this really sucks for the money I've paid for. I mean I can live with bugfixing. But it seems nothing will be done about this persistent loading problems.

Prusa-Support, 2022-07-15:
Thank you all for your feedback.

Could anyone reproduce the exact same problem described in the main post, to provide our developer @mkbel with a usable log? Maybe I missed that.

The explanation that @ANTALIFE provided some time ago is good input and the workaround looks promising.
@ANTALIFE Did you consider opening a pull request?
Sincerely though, I think that a slight movement of the top chimney toward the lever (to the left) would make the IR-lever triggering less sensitive to filament diameter changes.
Moreover, the IR-calibration should preferably be done with a 1.5 mm hex key to be forgiving enough in case of undersized (<1.75 mm) or too soft (squished) filament.
In the case of difficulties during the the top chimney adjustment, you may want to try chamfering the rim at the bottom of the "IR-holder", and gently sand the mating faces between "IR-holder" and "FS-cover". This could increase clearance and reduce friction for fine adjustments.

In the case of an irregular filament tip, the troubleshooting efforts should aim to avoid that. Most often a matter of temperatures and cooling moves. The filaments which we sell, if correctly stored, usually don't require tweaking. Our filaments could ease the troubleshooting.

Also, love the troubleshooting experience described by @lukasradek (this [comment](https://github.com/prusa3d/Prusa-Firmware/issues/2846#issuecomment-808736262)). It probably covers the most typical assembly issues. Very neat, congrats.

In most cases, similar issues could be caused by mixed luck with deformed filaments and ineffective IR-triggering.
This doesn't mean we can't work to improve firmware and designs.
In this case, logs would be appreciated  [#1689 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/1689#issuecomment-680850966).

Michele Moramarco
Prusa Research

traficwarden, 2022-07-22:
I can confirm with the original post with the ' load fail situation'.
I've also done a load to nozzle which works but will fail on the next load with the same error.
I don't think the mmu2 will ever work with just firmware updates. The complete unit needs a re design.
My MMU2 is still on my Prusa. Its days are numbered though if it cant be fixed. Many people have had hundreds of hours into trying to get theirs to work, just to have it brake down the next day with another problem. IS anyone listening from PRUSA?

mudkicker2, 2022-07-22:
@Prusa-Support 

I tried to get logs of this issue about a year ago.  I followed the instructions someone posted and was unsuccessful and never got any further help with that.  Honestly, at this point, I have chalked this up to a terrible investment with a company I will not do business with again.  I was really into the ideals behind this printer, but if I decide to go further into this hobby, I will go with a company that has a legit support structure.  Dealing with issues with this printer is ridiculous.  I feel as though these issues will be even harder to resolve as the newer products come.

Prusa-Support, 2022-07-27:
Thank you for the input.
A log would surely be appreciated but I totally get that clearer instruction would help. Maybe our content team can work on a tutorial - will see.

If anyone can reproduce the same issue described in the main post, but can't generate a log, please make a video and email it to my attention. Any evidence or visual documentation could hide some clues about the problem/bug too, as there may be mixed causes of the anomalies. A proper troubleshooting session could solve the case, reveal faulty parts, or confirm a bug.

_info@prusa3d.com
Att. Michele Moramarco
Please include a reference/link to this issue._

Michele Moramarco
Prusa Research

SeanI-Git, 2022-11-06:
I am having the exact same issue as the OP and many of the commenters have described. The MMU2S grabs the filament, loads it to the extruder, the IR sensor is triggered, and the MMU2S unloads again. It repeats 3 times and then stops. I have also been having the issue that other commenters have described, where at the end of every single MMU2S print, the filament unload cycle fails. I usually just have to press the knob and allow it to heat back up and try again, and it succeeds in unloading. 

I did not find this thread until tonight, and I thought it was just something wrong with my machine. I just spent all day giving my MK3S+ and MMU2S a complete maintenance overhaul, which was much needed by the way. And when this issue perstisted tonight, I hopped on and found out that I am not just crazy and apparently a lot of other people are having the exact same issue that I've been pulling my hair out over. 

I'll attach a link to a video I just made, with attempts 2 and 3 actually showing what the IR sensor is doing. I hope that this can be solved with a firmware update, and I think that it could be. I'm a bit surprised that with all of the issues people seem to experience with the MMU, the most recent firmware update was over 3 years ago. 

I will mention that after I made this video, I took Prusa Support's advice from the above comment from 7/15/22, and I adjusted my chimney (for the umpteenth time). The next print I tried started successfully, which was great! I hope it continues this way. I used a 1.5mm hex key and adjusted the chimney to the left until the IR sensor registered "0". Then I moved it ever so slightly back to the right until it registered "1". Hopefully this will make the load process more forgiving, as you can see in the video that whatever forces and filament bending that are going on in the extruder are presumably doing something to make the IR sensor read "0" for a split second, which, as another commenter mentioned above, is presumably what interrupts the load process. This definitely seems like something that could be fixed through a firmware update. Thank you Prusa Support, I really hope somebody gets on this issue.

[https://drive.google.com/file/d/1Xtn7mgoUfjYy7SddRRDhEdUsj5DLcYOW/view?usp=share_link](url)

MathieuGuennaoui, 2022-11-07:
> I will mention that after I made this video, I took Prusa Support's advice from the above comment from 7/15/22, and I adjusted my chimney (for the umpteenth time). The next print I tried started [...]

Thank you for that addition. I did the same a few months ago, and totally forgot to come back posting. My MMU is working flawlessly now, here are my modifications : 
- adjusted the chimney like [SeanI-Git](https://github.com/SeanI-Git) did ;
- ensured the path of the filament was as smooth as possible (slightly grinding some pieces to a better fit) ;
- designed a new support for MMU with variable inclination, meaning the ptfe tube linked to the extruder is less angled (will post it on Printables if someone needs it) ;
- printed a reinforced idler (a few actually, until I found the right version / parameters).

I didn't buy 3mm ptfe tubes for the 5x loadings but that might help getting less friction, which might help also. 
Printing different materials is still a pain, only because of the temperature management. For example, coming from PC to PLA will jam because the temperature goes down _before_ swapping. I'd rather get some overheated PLA push the remains of the PC out before letting it cool to printing temperature. 

[image]

Prusa-Support, 2022-11-12:
@sudo-gedit. Move the chimney towards the right to make it more forgiving (too much = IR permanently triggered).
Re-printing or post-processing the parts sometimes may help too. At least in case of deformed/worn parts, or poor clearance between the chimney parts to be adjusted.
If a filament un/load fails, likely Extruder-gears or MM-pulleys will become dirt and the filament will be dug and deformed. Clean gears and pulleys (careful with the idler tension screws) and cut out a large portion of filament in case it is deformed at the height of the MM-pulleys.

In case of irregular filament end, this article could help.
https://help.prusa3d.com/article/mmu-loading-failed_2135
Also, the filament should possibly avoid moisture.
https://help.prusa3d.com/article/drying-filament_332086

Get in touch with Tech Support for more help.
https://help.prusa3d.com/article/customer-support_2287

.
@MathieuGuennaoui. Thanks for the feedback but be careful with the PTFE tube angle. If too flat, it may be problematic when the Z-axis is in a high position.
The combination of materials is not much tested (we focused on supports material) and theoretically possible, but some plastics will simply chemically or physically not blend well together. A large change in temperature is a limit too as you already know. Feel free to share your outcomes, maybe in a separate issue or in a relevant topic of the [forum](https://forum.prusa3d.com/).

Michele Moramarco
Prusa Research

Wbmfishman, 2022-11-19:
guys I HAD the same problem. Went to Prusa chat and Pasted the post from linzwatt on 9-28-20.  No use trying to describe the issue since linzwatt explanation was perfect.  This is the link they sent me to and it worked. I had been doing a lot of 5 color prints. On the one were this issue started I gave up.  Did what they said in the link and boom reprinted the stl without issue.  Some how I think the code just gets corrupted over time and a complete erase of the epromm maybe after a few prints is a good idea to freshen things up. Give it a try and good luck.  https://help.prusa3d.com/guide/service-menu-factory-reset_86232.

wbmfishman

traficwarden, 2022-11-21:
> From "SeanI-Git" ***@***.***>
> To "prusa3d/Prusa-Firmware" ***@***.***>
> Cc "traficwarden" ***@***.***>; "Comment"
> Date 6/11/2022 3:53:22 PM
> Subject Re: [prusa3d/Prusa-Firmware] [BUG] MMU2S filament load and
> change fails 90% of the time (#2846)

I think the entire system is flawed. Prusa should redesign and forget 
having a tool change and be more like the Mosaic Pallet.
I've had my MMU2 on and off so many times hoping by a fluke it will work 
(never does and is always unreliable)
Very very disappointed and a waste of money.
Regards Ian

Prusa-Support, 2022-11-29:
A factory reset may rarely help - in the case of messed up values in the memory. This occurrence is usually hard to explain or foresee due to its glitchy nature.

Glitches are difficult to track down and reproduce.
Just to name a few possible causes of glitches: electrostatic interference, third-party devices, overlapping commands, bad wiring or cables, debris, manipulating the electronics while power is on, corrupted files/storage...

Glitches are rather seldom so a factory reset doesn't always help.
This advice may not specifically apply here on this issue, but, in general, anyone can give it a try.

.
The Mosaic Pallet is a very different product, it already exists and we have no interest in replicating it. It has many disadvantages but I don't find it fair as a Prusa employee to talk about this :slightly_smiling_face:
You can freely talk about Mosaic Pallet with the community on our [forum](https://forum.prusa3d.com/).
Please post about specific issues and feature requests here.

Nonetheless, our official Support can help you with the MMU2S, and rest assured that our FW developers are working on friendly usability, plus a few little but meaningful improvements.
The firmware is being rewritten, it may take more time but this will supposedly help in various ways.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-01-04:
Hi everyone. In case of anomalies, we have got instructions to generate a log now.
https://help.prusa3d.com/article/logging-data-over-serial-line-mmu2s_355360

Michele Moramarco
Prusa Research

Ic33s, 2023-01-04:
Hi, using the firmware from TheZeroBeast for MK3S + MMU2S it is nearly useable. Unfortunate, that the MMU2 FW update fixing the problems from prusa is still not available. (By the way: No hardware change to get it working at least half relyable.) The other firmware works faster, has better MMU2 status information ON SCREEN(!!!) and is a little more tolerant to what ever problem the prusa firmware has. I think if I have some time for some hardware modification on the selector side, I might get it working reliable. 

BUT now I rely on TheZeroBeast for updates. Very unfortunate for a commercial product.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg

lukasradek, 2023-07-31:
What was the issue?

gudnimg, 2023-07-31:
During development of the new firmware over the past year we've not seen this issue appear as far as I know. The MMU is loading much more reliably with the new firmware. We've improved a lot of things (make sure filament runout is impossible during loading/unloading, debouncing FINDA readout, completely re-write how errors are handled and recovered from, detect flickering FINDA sensor, and possible hardware issues for example with the stepper drivers etc.).

In my opinion, if users still see issues with the new firmware (3.13.0 and 3.0.0) then it would be best to create a new Github issue.

lukasradek, 2023-07-31:
Alright, we will see 🙂, sounds like good steps.

lukasradek, 2023-08-02:
@gudnimg 
Just curious... does the "rewritten from scratch" part apply to MMU2s fw as well?

gudnimg, 2023-08-02:
Yes :) The code was re-written on the printer side and entirely on the MMU side. We even have unit tests now on the MMU side 👍

gudnimg, 2023-08-03:
> Just commenting on this thread I rarely have a problem with PLA or PETG
since I have dialed in my Tool Change parameters and Ramm settings.  I have
2 MMU2s's. Waiting for upgrade kits. Will I be able to use the same setting
with the upgrade or have to start from scratch?

Hi, as far as I know the any Slicer settings which worked before should also work with the new firmware. I always default PrusaSlicer settings without any issues so far.

lukasradek, 2023-08-07:
> Yes :) The code was re-written on the printer side and entirely on the MMU side. We even have unit tests now on the MMU side 👍

I have been using it for a while now and have to say that is night and day. It sounds better, is moves better, it can prompt user via LCD.... just great!