prusa3d/Prusa-Firmware #2853: [BUG] Polish language does not work on 3.9.1 release

dlozinski, 2020-10-09:
On the first launch after flashing firmware version 3.9.1 MK3 (using PrusaSlicer 2.2.0+ on Mac) language choose menu pops up. Then if selected `Polski` language and confirmed printer restarts and pop ups the same menu again. Selecting `English` works.

leptun, 2020-10-09:
Do other languages work or are all of them broken except English? Also, please try flashing again and post here the logs from the flashing process.

dlozinski, 2020-10-10:
I had raspery Pi zero connected to the PI Port (running OctoPI). When I disabled PI port in the printes settings and flashed firmware again then I'm able to select `Polski` language.

dlozinski, 2020-10-10:
I'm sorry I cant flash it again and save the logs right now because the printer is currently very bussy :)

leptun, 2020-10-10:
I don't need logs anymore. What you said is enough for me. I know what went wrong since I am the one that created the bug apparently :)

leptun, 2020-10-10:
Nevermind. I handled this case as well, so there shouldn't be a problem flashing even if you have a raspberry pi connected. So if it is possible (no need to rush) please send me a log with how it fails.

Fredlandator, 2020-11-01:
I do have a similar issue with french language. 
after loading the firmware 3.9.1 using octoprint on a pi3+, firmware updater plug in, it asked me to choose a language. if french is chosen, it will then ask for a confirmation, and when confirmed , it reboots and ask for the choosen language again, and so on until you choose english.

samle if I unplug the pi from the mk3s.

I have no  idea about where to find logs, as it seems this may help help you fix this?

leptun, 2020-11-01:
@Fredlandator Languages only work after a firmware update only if you use PrusaSlicer for uploading the firmware. The Octoprint plugin for firmware update only flashes the main firmware, not also the languages

Fredlandator, 2020-11-01:
Ok, good to know that it doesn't work fully if not using prusa slicer.

After quickly double checking, it is not mentioned in the provided protocol that you HAVE to use slic3r PE. I feel it should. Or maybe I missed this information in the paper?

closed 2023-11-01 by github-actions[bot]