prusa3d/Prusa-Firmware #2854: [BUG]Errors during M860

andrewluebke, 2020-10-10:
**Printer type** - MK3S
**Printer firmware version**-  3.9.1

USB via Octoprint

**Describe the bug**
  For a couple of weeks when I compile the firmware myself and try to print something I no longer get "Recv: echo:busy: processing" messages instead I get "Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves."  This eventually leads to the print failing with a rerequested line error.  The stock firmware does not do this and send the busy messages.  I've tried completely erasing the build environment and rebuilding from scratch with the same issue.  Prior to this I had no issues with my self-compiled firmware.  The first file is what happens with my firmware and the second with the stock firmware.
  
**To Reproduce**
Build the current firmware, start to print anything.  No busy messages instead communication timeouts.
  
**Expected behavior**
Should get busy messages and no errors.
  
[No Busy.txt](https://github.com/prusa3d/Prusa-Firmware/files/5358258/No.Busy.txt)
[Busy.txt](https://github.com/prusa3d/Prusa-Firmware/files/5358263/Busy.txt)

leptun, 2020-10-10:
Can you please share a gcode or a PrusaSlicer profile that makes use of this M860 command? I want to make sure that I have a good base for the issue

andrewluebke, 2020-10-11:
I tried it with numerous models and profiles, but here is the gcode I tested to get the errors I reported.

[Full_0.2mm_PLA_MK3S_6h38m.zip](https://github.com/prusa3d/Prusa-Firmware/files/5360447/Full_0.2mm_PLA_MK3S_6h38m.zip)

andrewluebke, 2020-10-11:
This is the HEX file I generate when I compile.  I doubt whether it's a particular printer problem since if causes issues on all of my printers.

[prusa3d_fw_MK3S_3_9_1_3518.zip](https://github.com/prusa3d/Prusa-Firmware/files/5362247/prusa3d_fw_MK3S_3_9_1_3518.zip)

leptun, 2020-10-25:
I found what was wrong. A fix will come soon.

andrewluebke, 2020-10-30:
I can confirm that https://github.com/prusa3d/Prusa-Firmware/pull/2870 fixes my issue.  Thank you!

PR #2870 (2020-11-30): cmdqueue: Patches to Serial communication, pausing and file completion check
This PR makes a lot of small changes all around the code regarding the cmdqueue.
Things fixed so far:
1) If for some odd reason octoprint decides to get a file list during an SD print (happened to me while I was running the firmware in the emulator), then the current filename will change to something else. I patched it so even if the long filename was for some reason broken, it would still send back to octoprint at least some filename for it to parse correctly the M27 response.
2) Fixed bug introduced with #2832 . Now [... cut]
closed 2020-11-30 by DRracer