prusa3d/Prusa-Firmware #2856: [BUG] File Incomplete Errors opening files on SD Card post firmware 3.9.1 Final update

shofeur, 2020-10-11:
**Printer type** MK3S
**Printer firmware version**  3.9.1 Final

**SD card

**Describe the bug**
Following update to 3.9.1 I am experiencing two different failures when attempting to print from the SD card.

The first condition:

 When attempting to execute g-code files from the SD card the printer gives the error message, "File Incomplete. Continue anyway?" Selecting "Yes" will cause the printer to revert to the "Info" screen with the filename scrolling across the bottom but no printing activity occurs (temps remain 0, etc).  Selecting "No" will cause the printer to revert to the standard "Info" screen.  When a subsequent attempt is made following the "No" selection the printer firmware will reboot.  This behavior can be reproduced with the g-code in file Condition1.zip

The second condition:

Attempting to execute the g-code in file Condition2.zip causes the printer to return to the "Info" screen with the filename scrolling across the bottom of the screen but no g-code is executed.

This failure condition initially occurred with the Firmware 3,9.0 Final release.  Full factory reset has no effect on the problem.  Changing SD cards has no effect.  Reverting to Firmware 3,8.1 Final resolves the issue with no other changes or actions required.

**G-code**
[Condition1.zip](https://github.com/prusa3d/Prusa-Firmware/files/5360535/Condition1.zip)
[Condition2.zip](https://github.com/prusa3d/Prusa-Firmware/files/5360536/Condition2.zip)
 
**To Reproduce**
  See description above
  
**Expected behavior**
  Normal g-code execution from SD card.
 

Thank you,

Steve

Ruhkukah, 2020-10-13:
@shofeur 
This is a known bug, caused by scanning the printed file for the M84 command followed by carriage return. The firmware assumes that M84 is the last command in your file. Until the developers fix it in the future, you may run this file on FW 3.9.0 or 3.9.1 by modifying your Condition1 file: simply hit Enter button at the end of the line "M84 ;disable motors" and save the file.

shofeur, 2020-10-13:
Awesome, thank you for your reply.  I wish I had a better understanding of the search here so that I wouldn't have apparently duplicated a report.

Regards,
Steve

Ruhkukah, 2020-10-14:
Sorry, I should have used a different wording here. I meant "known bug to me" :)) I suggest that you keep open the issue so that the developers know it

shofeur, 2020-10-14:
Closed due to misunderstanding.  Bug report is still necessary.

s4eric, 2020-10-20:
I'm encountering the same issue on two separate printers; however editing the .gcode, as well as rolling back the firmware did not resolve it. Older .gcodes on my SD cards still print, but I cannot slice any new ones.

shofeur, 2020-10-20:
I am absolutely not any type of expert but I will note that Ruhkukah's recommendation must be exactly followed.  The M84 command must be the last command in the file and it must be followed by a blank line (i.e. the carriage-return must include a line-feed).  Every file that I edited in this way has worked flawlessly now when executed from the SD card.  Is your slicer adding custom g-code to the files? I had to edit my custom g-code to ensure that the M84 command was last and followed by a blank line.  Note that following comments appear to have no effect.

Regards,

Steve

Panayiotis-git, 2020-10-20:
Check [this issue](https://github.com/prusa3d/Prusa-Firmware/issues/2328) also.

s4eric, 2020-10-20:
Thanks for the suggestions, tried both with no change. 

I also tried to slice a file for the MINI and encountered similar issues. So I'm guessing it has to be related to PrusaSlicer. I tried re-installing to no avail. I have not added anything to the custom G-code sections, those sections remain locked as the default.

Ruhkukah, 2020-10-24:
@s4eric can you attach your gcode for us to check?

s4eric, 2020-10-25:
I figured out the issue, it was a combination of my SD card adapter and corporate policy blocking removable storage devices (work laptop). The .gcodes would save to the SD cards but not be able to be read by the printer. Switched to another SD card adapter and everything went back to normal.

closed 2023-11-01 by github-actions[bot]