prusa3d/Prusa-Firmware #2857: [BUG]<Enter comprehensive title>

vectordesigngroup, 2020-10-11:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3S]
**Printer firmware version**-  [3.9.1]

**MMU Upgrade** - [N/A]
**MMU upgrade firmware version [N/A]

**SD card or USB/Octoprint**
  SD Card

**Describe the bug**
  Printer was printing, and suddenly stopped half way through print job. It has happened 2 times on different models since the update. The printer just stops, and then you can see the os rebooting, then it does not show any resume options just stays there.
  
**To Reproduce**
  Can't reproduce sometimes it finishes, and sometimes it stops printing.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**

**Video**
  Please attach a video. It usually helps to solve the problem.

drelich, 2020-10-24:
Interesting. Also using 3.9.1 and just filed another BUG report 2868 because my printer also stops midprint randomly. But I print using OctoPrint. Still, this looks like both of us are having the same problem. Curious to see what'll be done about this.

vintagepc, 2020-11-02:
This sounds like a flaky SD card or card slot. Have you tried with a different one?

JohnnyDeer, 2021-08-19:
I'm closing this issue because there are no responses.

closed 2021-08-19 by JohnnyDeer