prusa3d/Prusa-Firmware #2858: [ENHANCEMENT] remind not to remove spring steel sheet for Z calibration

tehcaster, 2020-10-12:
After having moved the printer I ran Z celibration as suggested by the manual. As the initial calibration started with print sheet removed, I thought I should remove it also for this run. However, nozzle pushed into the heatbed during the second calibration point.

Stupid as I am, I managed to make the same mistake again after few weeks.

So in the name of dumb-proofing, I suggest that the message to check the nozzle for cleanness to proceed with Z calibration also reminded the user that the sheet should be kept installed, not removed. Thanks!

leptun, 2020-10-12:
Was this done with a MK2.5 or a MK3? The MK3 should stop after it pushes a bit into the bed and the damage should be minimal

tehcaster, 2020-10-12:
Sorry, forgot to say, it was MK3S. Edited. If I remember correctly it was: Point 1 - didn't push bed, Point 2 - pushed, continued, Point 3 - didn't push bed. Point 4 - pushed, stopped.

closed 2023-11-01 by github-actions[bot]