prusa3d/Prusa-Firmware #2860: duino IDE fails to compile with error  in EEProm M500.

bakerjvb, 2020-10-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3S]
**Printer firmware version**-(3.9.1 build 3518)

no mmu

**SD card

**Describe the bug**
  Arduino IDE failed to compile firmware 
Error was in eeprom.h.
static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500

error M500_conf is not a constant

To fix

I deleted "constexpr" making the line read
static M500_conf * EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500

and the Arduino IDE compiled the firmware correctly.

Tell me if I have made a mistake.

Rgds Julian Baker

3d-gussner, 2020-10-14:
You need to use Arduino IDE 1.8.5 please read the instructions carefully. Other option is to use PF-build.sh that will download and create a stable known build environment.

bakerjvb, 2020-10-15:
Anyone downloading Arduino will use the latest version and you will keep getting these errors. 
Here's an idea: change one line in the file.

leptun, 2020-10-15:
https://github.com/prusa3d/Prusa-Firmware/issues/2658#issuecomment-688201928

MikePfaff, 2020-11-30:
> [#2658 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/2658#issuecomment-688201928)

I know that changing line 594 in the eeprom.h from
`static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
to
`static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
fixes the problem for newer 1.8.x Arduino versions (tested on 1.8.12).

But the main question remains: **Does this one line change actually break things on Arduino 1.8.5?**

If not (= the changed line works also with Arduino 1.8.5), then i have to agree with bakerjvbs request:
> change one line in the file.

As far as i understood the cause and history of the problem, the line currently in the official source code is using some C++ constant casting magic which actually is invalid according to the C++ standard. Unfortunately, older compiler versions (like the one that came with Arduino 1.8.5) did not complain about this issue, so the problem never got noticed. Newer compiler versions (like the ones bundled with newer Arduino versions like 1.8.12) now will enforce the C++ standard, so suddenly the code won't compile anymore. So i don't get, why it's so hard to just change that one line from invalid to valid C++ code and be done with it.

gudnimg, 2021-12-10:
Fix for this issue was merged into 3.10.1 branch here: https://github.com/prusa3d/Prusa-Firmware/pull/2910 Arduino 1.8.5 is almost 5 years old now. I would think it's time to update, no? :)

closed 2023-11-01 by github-actions[bot]