prusa3d/Prusa-Firmware #2867: [FEATURE REQUEST]<Password for Settings>

HiWiFZI, 2020-10-23:
**Printer type** -  MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2

Problem:
Currently I'm running a Printer array of 6 Mk3s Printers at work. I often have the problem, that other not trained users are changing settings of the Printer, especially Z-Probe offset or crash detection. I constantly have to check and readjust those settings. 
  
Solution idea:
I would like to have to possibility to lock the settings menue to prevent mistakes. An Idea were using some kind of passwort to lock out users or a sort of combination. It could be possible to use some kind of interface like in a washing machine when using the "lock out feature for coin use". In Marlin 2.07 there is also the possibility to create a password to lock the settings.

DRracer, 2020-10-26:
This is a very unusual idea, however not completely out of the window.
For the 8-bit FW, implementing this may eat up a lot of resources, but we may consider this approach for the 32-bit FW.
If you want to physically lock the machines, you can design some kind of mechanical lock over the LCD/knob, may be even with a key lock. You may also consider configuring your machines remotely via serial port (Octoprint, Pronterface etc.).

HiWiFZI, 2020-10-26:
Having a physical lock is no possible solution in my case even if this would be a perfect way to solve the problem. Is there a possibility to disable double tap to babystepping? 
I understand, that for the 8bit variants it would take a lot of resources but for the 32bit it would be perfect.

closed 2023-11-02 by github-actions[bot]

nyckidryan, 2024-07-17:
Just wanted to second this idea... we have 13 printers in a maker space and there are very specific settings that we use to keep prints printing with a huge amount of reliability.