prusa3d/Prusa-Firmware #2868: [BUG] Printer sent 'start' while printing. External reset? Aborting job since printer lost state.

drelich, 2020-10-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S,
**Printer firmware version**-  3.9.1

**SD card or USB/Octoprint**
Printing via OctoPrint - latest build on RPi 3B+, direct GPIO to RPi port connection + external 2.5A 5V power supply.

**Describe the bug**
Print fails. OctoPrint log shows:
`2020-10-24 21:12:06,637 - octoprint.util.comm - WARNING - Received line:
2020-10-24 21:12:06,640 - octoprint.util.comm - WARNING - | \x00start
2020-10-24 21:12:06,646 - octoprint.util.comm - WARNING - The received line contains at least one null byte character at position 0, this hints at some data corruption going on
2020-10-24 21:12:06,651 - octoprint.util.comm - WARNING - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.`
 
**To Reproduce**
Cannot be reproduced. Issues seems coming at random.
  
**Expected behavior**
Complete a print.

I did find similar issue with previous versions of the firmware. Supposedly, the problem was the power panic detection that rebooted the printer. Since OctoPrint lost connection, it cancelled the print.

DaTTcz, 2020-10-25:
Hello. Are you using an external switch for the printer? I had the same problem. I was looking for a problem in the Octoprint, firmware, in the power supply and I had a faulty relay in the external switch. After replacing the relay, this problem is solved. I would try to replace the USB cable.

drelich, 2020-10-25:
Hi @DaTTcz,

I am using TP Link smart plug. It is as new as the printer itself (bought it only a week before my MK3S arrived earlier this month). The TP Link plugin shows no errors or reboots. It would also do its distinctive clicking when turning off/on, which I did not hear (the printer and the plug is only three feet away from me). The printer's screen did not flicker due to a power loss either.

Cheers.

leptun, 2020-10-25:
I've also had bad experiences with a smart plug before (not TP link). It would work fine for something like 30 minutes and then it would start generating resets every minute. I was printing from SD and the PowerPanic kept on resuming the print until I canceled it. Such a scenario during a USB print will 100% result in a print failure (and reset).
If you disable PowerPanic you might be able to print with the power plug. I'm not 100% sure, but I think if you disconnect the PowerPanic cable the printer will still work, but it won't reset anymore when power runs out.

DaTTcz, 2020-10-25:
> Hi @DaTTcz,
>
> I am using TP Link smart plug. It is as new as the printer itself (bought it only a week before my MK3S arrived earlier this month). The TP Link plugin shows no errors or reboots. It [...]

The status was the same for me. The switch looked fine, but it wasn't. Try a prints without TPLink.

drelich, 2020-10-25:
> The status was the same for me. The switch looked fine, but it wasn't. Try a prints without TPLink.

I think I will. I just had another failure :O despite the power panic cable out... Only this time, the toolhead would stay where it stopped (because of the panic cable out/power panic disabled).

drelich, 2020-10-25:
Actually, just found out the OctoPrint server was starting. My Raspberry Pi/OctoPrint box is not on any smart plug at all. It is directly connected to an outlet on the wall. I wonder do the vibrations loosen the power connector in the Raspberry Pi??? :O

DaTTcz, 2020-10-25:
Generally, remove all toys around the printer. Try printing from SD card only. If the problem does not occur, the problem is not with the printer and its firmware. Then add one toy at a time, switch... and try printing again.... octoprint... and try printing again...

drelich, 2020-10-25:
I think I'll do what you're suggesting, @DaTTcz, starting by removing the smart plug from the mix. Thanks!

spiff72, 2020-12-29:
I recently have been having similar issues.  Random resets of the printer at various stages of a print job.  My octoprint log looks basically the same as yours:

```
2020-12-28 20:09:15,137 - octoprint.util.comm - WARNING - Received line:
2020-12-28 20:09:15,138 - octoprint.util.comm - WARNING - | \x00start
2020-12-28 20:09:15,141 - octoprint.util.comm - WARNING - The received line contains at least one null byte character at position 0, this hints at some data corruption going on
2020-12-28 20:09:15,143 - octoprint.util.comm - WARNING - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
2020-12-28 20:09:15,148 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-12-28 20:09:15,154 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2020-12-28 20:09:15,154 - octoprint.events.fire - DEBUG - Firing event: PrintCancelling (Payload: {u'origin': u'local', u'path': u'strip_alignment_tool_122820b_0.2mm_PETG_MK3SMMU2S_2h3m.gcode', u'name': u'strip_alignment_tool_122820b_0.2mm_PETG_MK3SMMU2S_2h3m.gcode', u'owner': 'spiff72', u'size': 3923113L})
2020-12-28 20:09:15,155 - octoprint.filemanager.analysis - DEBUG - Resuming analyzer
2020-12-28 20:09:15,155 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xa3acd6d0>>
2020-12-28 20:09:15,159 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2020-12-28 20:09:15,161 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
```

I am using octoprint, running on a RasPi4 in a printed enclosure with a 7" capacitive touchscreen running OctoDash.  My Pi is connected via a homemade serial cable from the GPIO header to the connections on the Einsy board for the serial connection (where the Pi Zero would normally attach).  I don't have the 5v line connected between the Pi and Einsy - just ground and the other wires (Rx/Tx and two others that I don't recall the names of - one was a reset line I think?).

I find that it most often happens when I am physically interacting with something on/near my printer.  Today, I had just opened the enclosure door (lack enclosure) to tweeze a small strand of filament off the purge line before it circled back and  picked it up on the second pass over that line.  I think it happened as my hand neared the head, and maybe my forearm was near the LCD housing.   Another time it happened as I was reaching above my octopi server to unplug a USB cable from behind the enclosure (not connected to the Pi - it was plugged into a USB port on a UPS that sits behind the Pi enclosure.  

I am wondering if this may be an ESD related phenomenon?  I was just handling some parts right off the bed, and parts often seem to have a static charge buildup on them after I pull them off the bed.

spiff72, 2020-12-29:
Also, I have now enabled the serial log just in case that may be useful for diagnostics if it happens again.

spiff72, 2021-01-02:
I just had another incident, and this time I am SURE it was an ESD problem.  Print started, the MMU was having trouble getting the extruder to grab the end of the filament, so I reached in, and touched the tube fitting on the top of the extruder to push it in and pull the tube out before it gave up and retracted.  As soon as i touched the metal fitting - instant reset of printer.

It is the cold dry season here in Michigan - so static charges do tend to build up a lot.  I had just removed a large part off the bed and was cleaning it up when this all happened.  

Is there any way to prevent this other than making sure to touch a metal object before touching the printer?

leptun, 2021-01-02:
@spiff72 Do you have by any chance the PSU separated from the frame (as it is done with the PrusaBox enclosure as to not overheat the PSU)? When I had it like this on a silver PSU a year ago, every time I would sit down at the printer and touch the LCD I would get a really powerful ESD. It didn't reset the printer in my case, but it was enough to get the character LCD out of sync. Make sure the printer frame is grounded properly using the PSU's external casing (also check for faulty power extender with ground unplugged if you are using one).

spiff72, 2021-01-03:
> @spiff72 Do you have by any chance the PSU separated from the frame (as it is done with the PrusaBox enclosure as to not overheat the PSU)? When I had it like this on a silver PSU a year ago, every [...]

Yes indeed - I do have the PSU sitting on the desk behind my enclosure, but I have the black PSU (not silver).  I have also seen a garbled LCD at times.

What is the recommended way to ground the PSU to the frame?  Run a wire from some point on the frame (at a glance, I could probably rig up a wire and ring terminal to one of the screws that hold the frame brace (the printed part I grabbed to use in the location where the PSU would normally be), and I assume I would attach the other end to one of the tapped holes on the PSU housing (with a very short screw to make sure it doesn't go in too far)?

Also, I am not sure what you mean by "check for faulty power extender with ground unplugged if you are using one"?  What do you mean by "power extender"?  An extension cord?

Thanks!

leptun, 2021-01-03:
@spiff72 In my setup I just connected the ground wire that goes to the mainboard also to the frame. This might not work on the black PSU however. I just checked and with an unmodified MK3S the ground from the PSU is not connected directly to the PSU case. So it would be best to do it as you just said with a wire from the PSU casing to the frame. As for the screw, check how much it normally goes in by inserting the screw in the printer frame with no PSU and then use the screw that matches the size you found. Maybe you could also add some washers uf you think it’s a bit too long.

Yes, I meant to say extension cords.

spiff72, 2021-01-03:
Got it - thanks!

I will tackle this tomorrow.  Been busy all day with other projects and I am exhausted ;-)

spiff72, 2021-01-03:
I may try using one of these screw locations instead of the ones that would normally mount to the frame, since the frame screw locations are on the surface of the PSU that I have resting on the desk.  Otherwise, I think there are a couple of screw locations on the large flat face too - but the ones circled in red and green would be more convenient since they are right by the other wires that have to get routed into the enclosure.
[image]

spiff72, 2021-01-03:
I went ahead and connected this today.  I used the lower screw on the frame where the PSU attatches (pictured), and used the slot circled in red in the previous post for the connection at the PSU.  I also scraped away the finish around these connections to make sure it got good contact.

It was not printing when I started this, and when I reached in to loosen the screw on the printer frame, it reset!  Hopefully I don't see this happen as often now.

[image]

Thanks

JohnnyDeer, 2021-07-20:
@drelich is this issue still actual?

drelich, 2021-07-20:
> @drelich is this issue still actual?

I have found out that the problem was actually caused by the smart plug I was using. Since I removed it from the mix, I have not had a single failure.

JohnnyDeer, 2021-07-20:
@drelich thank you for info, so I'm closing this issue.

closed 2021-07-20 by JohnnyDeer