prusa3d/Prusa-Firmware #2869: [QUESTION] Why does the heating bed need to heat when I want to load filament?

ing-kuczmik, 2020-10-25:
When the printer is cold and I want to load or unload filament. Why does the heating bed need to heat when I want to load filament?
Best regards
Adam

neophyl, 2020-10-26:
Because you are loading filament the printer makes the assumption that you want to start printing after loading filament.  As it takes longer to heat the bed up this assumption is good for the majority of cases and decreases wait time for that majority of users.
It was probably done that way due to user requests.
If all you want is to do is load filament then you could always navigate to the settings screen and set the extruder temperature to a suitable value, wait till it heats up and then load filament.

thehumanfallacy, 2021-01-16:
This makes sense as I also usually want to print after loading.  But maybe it could also prompt and ask you if you want to heat the bed?  That should be a quick user interaction if the default value of _Yes_ is highlighted.  Just a thought.

Panayiotis-git, 2021-01-16:
You can find an ongoing discussion about this subject [here](https://github.com/prusa3d/Prusa-Firmware/pull/2335)

Prusa-Support, 2022-05-19:
merge with #422

closed 2022-05-19 by Prusa-Support