prusa3d/Prusa-Firmware #2876: [ENHANCEMENT] Should align Z axes to a fullstep when idling/poweroff

wavexx, 2020-10-29:
**Printer type** - MK3S, MK3 (partially MK2.5S and earlier)
**MMU Upgrade** - N/A

**Is your enhancement related to a problem? Please describe.**
We should ensure the Z motors are disengaged in a full step, when possible.

If the motors happen to be halfway between a fullstep, there's a good chance that two motors will snap to opposing fullsteps, skewing the axis. This is not usually a problem, since a single fullstep of difference with the leadscrew doesn't lead to significant changes (and these changes get compensated by the MBL anyway), however if this keeps happening it will create additional strain on the POM nuts, eventually leading to backlash.
 
**Describe the solution you'd like**
The printer should re-align Z to a fullstep before power off Z when possible and sensible to do so.
Note that this is already implemented when PowerPanic kicks in, in order to preserve an accurate position after the power-cycle. It just needs to be re-used.

On the MK3S we can detected when the power is switced off by the button. In this case, and if the Z motors are still powered, alignment should be performed. On the MK3 this is only possible if the power chord is pulled, but the same reasoning applies.

It's harder to say what should be done with a M84 (release steppers) and with the inactivity shutdown timer (M85), since this causes an unexpected move which is not requested. I feel that M84 should release the steppers immediately, but M85 could be allowed to re-align all axes before release, as long as we return to the original position when powering them on again.

A good interim solution IMHO would be to add a new M-code to perform the stepper alignment. Such code should be called by the printer end-gcode section just prior to releasing the steppers. This is simple, easy, and avoids skew build-up in the most common case.

Prusa-Support, 2023-05-10:
Hello. Probably this is nothing to be worried about but it may be worth mentioning issue https://github.com/prusa3d/Prusa-Firmware/issues/4019. The user was concerned by this behavior because the Z stepper motors were changing position very often.

This almost random Z movement seems negligible but I can't exclude edge cases where such erratic behavior becomes relevant.

Michele Moramarco
Prusa Research

leptun, 2023-05-13:
> A good interim solution IMHO would be to add a new M-code to perform the stepper alignment. Such code should be called by the printer end-gcode section just prior to releasing the steppers. This is simple, easy, and avoids skew build-up in the most common case.

@wavexx I agree with this. I am not really fond of strapping this Z alignment functionality onto an existing gcode command. It would just add confusion just like the Delta_PSU code did. So I propose we add another gcode command just for this and have the slicer scripts updated to account for this. Do you know if there is any gcode already reserved in some other firmware for such an action?

closed 2024-10-16 by github-actions[bot]