prusa3d/Prusa-Firmware #2879: [BUG] MINI touched and bended hotend during Mesh Bed Leveling calibration

jakubmirejovsky, 2020-10-31:
Prusa MINI, firmware 4.2.1, no mods

Right after Mesh Bed Leveling (under Calibration), the nozzle goes down and touches and bends the hotend.
Video here: https://www.youtube.com/watch?v=4gR8gOi_ESg
  
**To Reproduce**
It happended to me when I:
- started printing (went for 15 minutes)
- paused printing for 6 hours
- resumed printing
- stopped printing after 5 minutes
- Z axis moved up
- first layer calibration
- Mesh Bed Leveling - here the problem was spotted. I tried it again with my camera on (see youtube link).
  
**Expected behavior**
I expect nozzle not to touch the surface at all.
  
**G-code**
I am attaching the gcode, but I don't think it will be useful for debugging:
https://www.dropbox.com/s/63aloo0s4qaw169/Spi_FRA_public_up_cov_0.25mm_PLA_MINI_3h27m.gcode?dl=0

**Video**
https://www.youtube.com/watch?v=4gR8gOi_ESg (0:43)

vintagepc, 2020-11-02:
This github is for the MK3 firmware. You are looking for the Mini firmware repo:

https://github.com/prusa3d/Prusa-Firmware-Buddy

Please file your issue there.

jakubmirejovsky, 2020-11-09:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/981

closed 2020-11-09 by jakubmirejovsky