prusa3d/Prusa-Firmware #2881: [BUG] Wrong filament sensor detection message

rongith, 2020-11-01:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]
MK3S 3.9.1

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]
No MMU

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint
Not relevant

**Describe the bug**
Finally I found that my filament sensor is good although when I wanted to enable it I got the message "ERROR: Filament sensor is not responding, please check connection." As I didn't took time to try to make it work, it was disabled for few monthes.
  
**To Reproduce**
Remove filament
Switch off printer
Switch on printer
Check to enable filament sensor in settings, you see N/A. If you try to enable it, you got "ERROR: Filament sensor is not responding, please check connection."
Insert filament to trigger the sensor.
Check to enable filament sensor in settings, you see N/A. If you try to enable it, you succeed.
  
**Expected behavior**
If the filament sensor is good but there is no filement, we shouldn't have the message "ERROR: Filament sensor is not responding, please check connection."
The message has to be changed (laddind a "or there is no filament inserted, insert one to enable the sensor"...) or the filament sensor detection procedure has to be changed.

leptun, 2020-11-04:
Are you sure you didn't accidentally install the MK3 (not S) firmware on a MK3S printer?

rongith, 2020-11-04:
Nope, I just checked in my dowloaded files. It is prusa3d_fw_MK3S_3_9_1_3518.
I issued M115:
```
Send: M115
Recv: FIRMWARE_NAME:Prusa-Firmware 3.9.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
```

IR sensor status is correct in "Support"/"Sensor Info" and voltage is correct too between GND, Vin and Data when i insert or remove a filament.

3d-gussner, 2020-11-07:
@rongith Can you please check which IR sensor has been detected and which Voltage levels are shown?
- Support -> "Fil. sensor v.:"
- Support -> Voltages -> "IR:"
  - After startup without filament
  - with filament
  - without filament

Do you know which hardware IR sensor version do you have? Is it an original IR sensor, is it a modified one (like LED mod)?
See also https://help.prusa3d.com/en/article/ir-filament-sensor-detection-mk3s_159426

Please run Settings -> HW Setup -> Fsensor Detection

If you have a chance to record the IR sens pin voltages during power up, insert and remove filament it could help to troubleshoot the issue.

My attempt to reproduce this issue:
- I tried all kind of variants of filament loaded/unloaded, power off/on, wiggle the filament and could not force the it.
- BUT I could force IR sensor to FAIL (N/A) by getting it to "unknown" IR sensor state by setting the voltage of the sens pin to ~3V

The message "ERROR: Filament sensor is not responding, please check connection" should only be shown if the IR sensor readings are out of allowed ranges.

rongith, 2020-11-09:
@3d-gussner 3d-gussner
> Can you please check which IR sensor has been detected and which Voltage levels are shown?
>      * Support -> "Fil. sensor v.:"

After power up without filament: 0.4 or newer
After inserting filament: 0.3 or older

After power up with filament: 0.4 or newer
After removing filament: 0.3 or older

>     * Support -> Voltages -> "IR:"
>       * After startup without filament
5.0V
>       * with filament
0.1V
>       * without filament
5.0V

> Do you know which hardware IR sensor version do you have? Is it an original IR sensor, is it a modified one (like LED mod)?
> See also https://help.prusa3d.com/en/article/ir-filament-sensor-detection-mk3s_159426

I know there were a newer filament sensor. But not a new feature about detection for 0.4. Mine is 0.1, an original Prusa one.

> Please run Settings -> HW Setup -> Fsensor Detection

After following the procedure, I got "sensor verified".

> If you have a chance to record the IR sens pin voltages during power up, insert and remove filament it could help to troubleshoot the issue.

No I have no tool to record the voltage during those steps.

> My attempt to reproduce this issue:
>     * I tried all kind of variants of filament loaded/unloaded, power off/on, wiggle the filament and could not force the it.

Thanx to investigate

>     * BUT I could force IR sensor to FAIL (N/A) by getting it to "unknown" IR sensor state by setting the voltage of the sens pin to ~3V

:thinking:
Should be something in the power up giving wrong indication on the filament sensor version.

closed 2023-11-02 by github-actions[bot]