prusa3d/Prusa-Firmware #2883: [BUG]Crash Detected, Nozzle collides with Model in back left corner only

EasyAsABC123, 2020-11-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.1-3518

**MMU Upgrade** - N/A
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint** Octopi
```
2020-10-30 03:14:26,734 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 6385, current line = 6395
| Last lines in terminal:
| Send: N6394 G1 F8640.000*98
| Recv: echo:enqueing "G1 E-1.000 F2700"
| Recv: echo:enqueing "G1 Z10.492 F800.000"
| Recv: echo:enqueing "CRASH_DETECTEDX"
| Recv: tmc2130_home_enter(axes_mask=0x01)
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv:   0 step=14 mscnt= 227
| Recv: tmc2130_goto_step 0 45 2 1000
| Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
| Recv: tmc2130_home_enter(axes_mask=0x02)
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv:   0 step= 1 mscnt=  31
| Recv: tmc2130_goto_step 1 1 2 1000
| Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
| Recv: echo:enqueing "CRASH_RECOVER"
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: Resend: 6385
2020-10-30 03:14:48,473 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 6429, current line = 6446
| Last lines in terminal:
| Send: N6445 G1 F1500.000*103
| Recv: echo:enqueing "G1 E-1.000 F2700"
| Recv: echo:enqueing "G1 Z10.501 F800.000"
| Recv: echo:enqueing "CRASH_DETECTEDY"
| Recv: tmc2130_home_enter(axes_mask=0x01)
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv:   0 step=46 mscnt= 738
| Recv: tmc2130_goto_step 0 45 2 1000
| Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
| Recv: tmc2130_home_enter(axes_mask=0x02)
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv:   0 step= 1 mscnt=  29
| Recv: tmc2130_goto_step 1 1 2 1000
| Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
| Recv: echo:enqueing "CRASH_RECOVER"
| Recv: echo:busy: processing
| Recv: echo:busy: processing
| Recv: Resend: 6429
```

**Describe the bug**
After upgrading to 3.9.1, prints fail if the build plate back left corner is used.  The nozzle crashes into the model, i've rerun calibration and setup wizard with the same results.
  
**To Reproduce**
Print anything in the back left corner
  
**Expected behavior**
Printing should succeed without crashes
  
**G-code**
[Link]<https://privatebin.net/?a001e7764a43afea#6dX1idQQq8Yey3byhNxWCDpPT16QpPYN7bSjCFa14rUu>

**Video**
None to attach, if desired I will record and upload.

kozmaz87, 2021-04-04:
I can only say that there is no crash really it is a false alarm and for me it is when I am putting the model in the front left quadrant close to home position and whenever the nozzle is around the back right part of the model it misreports crash. I had to turn crash detection off.

Presswerk65, 2021-04-05:
I had the exactly the same problem - have you checked your zip ties? In my case the zip ties for the cables to the hotend touched the head bed connector cover in the left rear end

kozmaz87, 2021-04-05:
That is not the same then... you had an actual crash... we did not.

arkovda, 2021-07-20:
I have had 2 crashes with the same gcode (created using PrusaSlicer 2.3.0)

Printer type - MK3S
Printer firmware version- 3.10.0

MMU Upgrade - N/A
MMU upgrade firmware version - N/A

SD card or USB/Octoprint Octopi

[octoprint.log](https://github.com/prusa3d/Prusa-Firmware/files/6850092/octoprint.log)

leptun, 2021-07-20:
@wavexx Can we dump all the info we can from the tmc drivers when the crash occurs? I'm curious if somehow the driver resets or some parameter gets changed unintentionally (best I can think of are GCONF and TCOOLTHRS changing unintentionally).

kozmaz87, 2021-07-20:
YESSS people can reproduce it... we are at step 3 :D 
[image]

arkovda, 2021-07-21:
Thanks for quick answer.
Sorry but I have overwritten the bad gcode with a new version and that worked fine.
I have tried to reproduce it using the same object. But no luck. Not much more log to find unless you can tell me exactly where to look. I will try harder to reproduce the problem again.

arkovda, 2021-07-21:
I have an idea. 
I have filament leakage and this might cause the problem.

leptun, 2021-07-21:
@arkovda yes, that can be the cause. If the filament is leaking, then the nozzle and the heatbreak aren’t flush with one another. This means that the nozzle and the heater block can move a bit. This can be extremely problematic when printing infill as the nozzle will probably hit the previous layer’s infill and cause a crash to register or at the very least some vibrations.

arkovda, 2021-07-21:
Ok, thanks 
For now I'm happy with that.
I will try to sort the leakage.

closed 2023-11-02 by github-actions[bot]