prusa3d/Prusa-Firmware #2886: [ENHANCEMENT]<Enter comprehensive title>

GWdd, 2020-11-06:
**Printer type** - MK3, MK3S, ... 

**Is your enhancement related to a problem? Please describe.**
When updating firmware, too often something happens and NVRAM is corrupted. This requires a full factory reset with data clear and a subsequent full printer recalibration and manual setting of ALL use adjustments like bed correction, linearity correction, PID values, z-offsets per sheet, etc. A lot of time required that needn't be.

Plus, if a person misses checking each and every setting available via menus, some settings are lost and must be relearned (usually requiring hours of multiple test prints).

**Describe the solution you'd like**
Modify M503 to provide a COMPLETE dump of all NVRAM user settings. 
 -- It would be extremely nice to also provide a reciprocal to M503 that loads and stores that data back to NVRAM.

M503 S1 ; save everything a use can set
M503 L[all the stuff dumped from S1] ; load settings back into printer

Lack of a way to easily determine what is in memory is almost a defect/bug. So I don't think this is an unreasonable request, and simply including everything in the dump should not add significantly to the already short supply memory space. 

I'd even be happy with simple values by address so I can compare before a FW update vs after a FW update

M503 Sx
4095 = 0xD7h
4094 = 0x09h
...
etc

Adding a way to load settings is new, so may not be easy to do with current constraints. But it would be nice to have.

3d-gussner, 2020-11-07:
Thanks for the request.
Please checkout https://help.prusa3d.com/en/article/prusa-specific-g-codes_112173#d-codes `D3` gcode command
and see https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/eeprom.h#L42-L378 which EEPROM address is what and which values are allowed.

- G-code `D3` without any parameter will dump the EEPROM.

**Please be careful with modifying the EEPROM directly as it might damage your printer!**

Writing with `M503` wouldn't be the correct g-code as it is used to "Report current settings" see https://reprap.org/wiki/G-code#M503:_Report_Current_Settings

I hope this is an exactable workflow for you to be able to dump the EEPROM and compare different versions, make a backup, or ....

BTW: Next time please do not forget to edit the Title in the issue template 😉

GWdd, 2020-11-07:
D might work, though the ideal is to have a commented or verbose report of settings that is human readable rather than a simple hex dump.  The dump will get me by, but ... lol.

I think I'll keep the request current rather than close it, as it's still a nice feature to have.

leptun, 2020-11-07:
There are not enough resources for a more detailed eeprom dump. The most we can get is a hex dump, but not something where each and every value is explained. For this purpse I created myself a template in a program where I can parse a hex/bin file of the eeprom and it returns me a parsed list of parameters. I’ll write more about this if there is interest, but keep in mind that the program I used is trial only unfortunately.

GWdd, 2020-11-07:
What is a bit disturbing, and problematic, is that M503 says it is dumping all user settings. But it isn't.  If resources are so slim the command can't work properly, document it as such, or somehow fix it.  
Instead of a partial verbose dump, dump what you can verbosely, then dump the rest of the user settings and just say "other" - so users know there are other settings. 

And I don't need to read "Bed calibration constant front" "Bed calibration constant left" "Bed calibration constant right" "Bed calibration constant rear" : "Bed Level" 50 25 30 0 is enough detail to be fully useful. This is true of all of the settings being dumped. Even "Bed Comp" 1 "Bed Level" 50 25 30 0 is enough context to know bed comp is enabled and what the settings are.

Here are the actual dumps: the amount of descriptive text is helpful, but totally unnecessary:
echo:**UStep resolution**: _uStep res_ or _M350_
echo: M350 X16 Y16 Z16 E32
echo:Maximum feedrates - normal (mm/s): _M203_
echo: M203 X200.00 Y200.00 Z12.00 E120.00
echo:**Maximum feedrates - stealth (mm/s):** _M203 stealth_
echo: M203 X100.00 Y100.00 Z12.00 E120.00
echo:Maximum acceleration - normal (mm/s2): _Max accel_ or _M201_
echo: M201 X1000 Y1000 Z200 E5000
echo:Maximum acceleration - stealth (mm/s2): _stealth_
echo: M201 X960 Y960 Z200 E5000
echo:**Acceleration: S=acceleration, T=retract acceleration** _M204_
echo: M204 S1250.00 T1250.00 
echo:**Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s), Z=maximum Z jerk (mm/s), E=maximum E jerk (mm/s)**  _M205_
echo: M205 S0.00 T0.00 B0.00 X10.00 Y10.00 Z0.40 E2.50

Lots of opportunity to save space here. But sure, the code to read the bytes as groups takes up space too, but I'd think small compared to even one line of 180 characters of text in multiple languages.

GWdd, 2020-11-07:
@3d-gussner 
How does one actually use the D command? The links say it reads/writes, but doesn't really explain the difference between reading vs writing.

D3   
by itself I think reads everything, but what about reading just a single address?

D3 A100 
does this read one byte from 0x100? Or does it write a nul to that address?

D3 A100 C1 FF 
should write a single byte... but I am mostly guessing.

3d-gussner, 2020-11-07:
@GWdd Here the part describing the 'D3` gcode https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Dcodes.cpp#L103-L120

also in the https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/eeprom.h#L42-L378 you can see in the "Debug code" tab.

Example:
To read `EEPROM_FAN_CHECK_ENABLED` use `D3 Ax0f87 C1` which translates to
D3 = Read/Write EEPROM
A   = Address `Ax0f87` (**!!! The hex address needs to be lowercase without the 0 before the x)**
C   = Count `1` read one byte

Result when Fan check is ON
```
Send: D3 Ax0f87 C1
Recv: D3 - Read/Write EEPROM
Recv: 0f87  ff
Recv: ok
```
Change the Settings -> Fan check to Off via LCD will give the result
```
Send: D3 Ax0f87 C1
Recv: D3 - Read/Write EEPROM
Recv: 0f87  00
Recv: ok
```

Send: `D3 AAx0f87 Xff` and the Settings -> Fan check will be back On after printer restarts.

Some write changes are visible immediately others later or after restart of the printer.

Hope that helps.

As @leptun already wrote there is limited space and changing the M503 output as you describe can cause more unexpected issues with host systems and other things.

GWdd, 2020-11-07:
Perfect - thank you.

GWdd, 2020-11-24:
...

closed 2020-11-24 by GWdd