prusa3d/Prusa-Firmware #2887: [QUESTION] XYZ Calibration on custom sized prusa

evilguy422, 2020-11-09:
Hello,
I build a custom sized prusa using prusa hardware. I compiled a custom firmware where i put the new build volume into the configuration_prusa.h file. The size does seem to work when I manually move each axis through the printer, however when I run an XYZ calibration it tests the X axis and says it failed and to check if the pulley is loose. Any idea what I can do to fix this? I quickly looked through the firmware files but I am really not familiar with code so I cant tell if I need to edit anything else. Any pointers would be great, thank you

3d-gussner, 2020-11-09:
@evilguy422 Thanks for the question. Bit more information what "custom sized" means would be helpful as there are different locations you might to have a look depending on what you have changed.
- x-axis = ???
- y-axis = ???
- y-axis = ???
- Extruder = did you change anything here?
  - Sensor location and which one do you use?
- Which bed do you use?
  - Does the bed has calibration points?
- more details

evilguy422, 2020-11-09:
Hello @3d-gussner Thank you for the reply. I will try to give as much detail as possible.

> * x-axis = 306mm
> * y-axis = 312mm
> * z-axis = 359mm
> * Extruder = default prusa mk3 (not s) extruder with pinda and the filament sensor. The only thing I changed is the heater block is not a volcano and to help with the cooling I added a second part cooling fan.

>   * Sensor location and which one do you use?
> * Which bed do you use?
Its just a regular square heater bed off of aliexpress, as nothing like the prusa bed exists in a large size from what I found.

>   * Does the bed has calibration points?
This was going to be my follow up question. There are no points. However from what I understand, the pinda probe works by detecting change with magnetic field or something like that correct? The original bed is aluminum which isnt magnetic, and where the circles are on the bed for the calibration, there are no magnets. The magnets are a few cm away from those calibration circles. I can always glue some magnets on my bed if thats how it work, but I dont know where to position them as these points dont have magnets.

Other than that, the frame is based off the bear upgrade. The bed thermistor is original prusa one. The steppers arent all original. The Z ones work well, and the extruder seems fine, but the cheap steppers didnt work well with the trinamic for x and y. I coudlnt dial in the correct current for them to detect the end stop while homing so for now I switched to original prusa motors for x and y. Let me know if any other info would help. Thanks

evilguy422, 2020-11-11:
Anyone?

3d-gussner, 2020-11-16:
@evilguy422 Sorry for the late answer.
As you have modified quite a lot I would suggest that you to compile Marlin 2.0.x. or try get some help in other custom firmware repos/forks, like: https://github.com/vertigo235/Prusa-Firmware

As a side project I started https://github.com/3d-gussner/Marlin/tree/bugfix-2.0.x_MK3s_Einsy but it is WIP and needs more attention. Maybe this is a base you can work on and get help from the Marlin community.

As there was no reaction from other community members I guess you are on your "own". Sorry.

Please understand that this is a highly modified printer and Prusa may not being able to help you with your issue.

> This was going to be my follow up question. There are no points. However from what I understand, the pinda probe works by detecting change with magnetic field or something like that correct? The [...]

You can see here https://github.com/prusa3d/Heatbed_MK52_magnetic that there are 4 calibration points on the MK52 bed (9 on MK42) which are part of the PCB.
The Prusa-Firmware looks for these points with the PINDA to measure the skew of the printer.
Which looks bit like this:
```
Recv: xyzcal_find_bed_induction_sensor_point_xy x=1200 y=600 z=2000
Recv: point=0 x=1200 y=600 z=2000
Recv: xyzcal_meassure_enter
Recv: xyzcal_searchZ x=1200 y=600 z=2000
Recv: xyzcal_spiral8 cx=1200 cy=600 z0=2000 dz=100 radius=900 ad=0
Recv: xyzcal_spiral2 cx=1200 cy=600 z0=2000 dz=100 radius=900 ad=0
Recv:  ON-SIGNAL at x=1115 y=295 z=1964 ad=274
Recv: sizeof(block_buffer)=1760
Recv: sizeof(total)=1080
Recv: xyzcal_scan_pixels_32x32 cx=1200 cy=600 min_z=1892 max_z=2400
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000050c00000000000000000000000000000000
Recv: 0000000000000000000000000a242f2e200b0000000000000000000000000000
Recv: 000000000000000000000001133549514b391800000000000000000000000000
Recv: 0000000000000000000000163f58646254391200000000000000000000000000
Recv: 0000000000000000000000072d4e646b66543407000000000000000000000000
Recv: 0000000000000000000000163f58646254391200000000000000000000000000
Recv: 000000000000000000000001133549514b391800000000000000000000000000
Recv: 0000000000000000000000000a242f2e200b0000000000000000000000000000
Recv: 0000000000000000000000000000050c00000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv: 0000000000000000000000000000000000000000000000000000000000000000
Recv:   0 979
Recv:   1 8
Recv:   2 9
Recv:   3 9
Recv:   4 5
Recv:   5 7
Recv:   6 7
Recv:   7 0
Recv:   8 0
Recv:   9 0
Recv:  10 0
Recv:  11 0
Recv:  12 0
Recv:  13 0
Recv:  14 0
Recv:  15 0
Recv: max_c= 9 max_l=2
Recv:  min=0 max=128 dif=128
```

So adding some magnets will not help you with this issue.

DRracer, 2020-11-20:
@evilguy422 I'm afraid you ran into a serious problem - the XYZ calibration algorithm has been tuned for Original Prusa Printers, most notably for the HW components we use. The algorithm is very sensitive to HW components and it is likely it won't survive changes in printer's dimensions and/or bed calibration point changes.
Unless you dig into the code and find out yourself what is failing and why, we cannot help you very much.

evilguy422, 2020-11-20:
@3d-gussner @DRracer 
Thanks for the info guys. Didn't realize there was something in the bed. I guess my last question would be, is it possible to run a custom marlin firmware on the Prusa Rambo or would I need to get a new board for this printer?

3d-gussner, 2020-11-20:
It is possible to get vanilla marlin working on Einsy board.
I posted a link above to my github marlin fork, this could be a starting point for you. As said before it is WIP so not fine tuned.

nosRider1, 2020-12-19:
I used a 0.9 degree motor and suffered from the same problem.
The solution is to tighten the belt more.
Tighten it little by little and check it with the Belt Test.
I solved it that way.

closed 2023-11-02 by github-actions[bot]