prusa3d/Prusa-Firmware #2889: [BUG] M602 does not resume after PRINT FAN ERR condition

jcwren, 2020-11-11:
**Printer type** - MK3
**Printer firmware version**-  3.9.0

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
Recently I started getting sporadic PRINT FAN ERR messages, even though the fan is spinning. I can resume the print from the front panel, but sending M602 from the Octoprint terminal does not cause the print to resume.
  
**To Reproduce**
Induce a PRINT FAN ERR, send M602 from Octoprint terminal.
  
**Expected behavior**
I would expect the print to resume as if I had selected Resume Print from the front panel

3d-gussner, 2020-11-16:
@jcwren Thanks for the bug report.
I tried to reproduce this bug, but as you wrote it is sporadic, and I couldn't get it to fail.
But I have seen that once in the past. I am not sure why it happened and that it is related to `M601/M602`.

There is a PR https://github.com/prusa3d/Prusa-Firmware/pull/2870 improving several issues (one is `M601/M602`) and maybe it also fixes this sporadic error.

If **you** are able to reproduce this error please make a short video.

leptun, 2020-11-16:
@3d-gussner @jcwren This issue will be present also with @2870. Unfortunately, right now I can't make it work as you want without causing issues with other host software (let's say Pronterface). I need a way to differentiate between octoprint and other more simple host software.
Right now, during a fan error the serial stream is forcibly paused by the printer since Pronterface and such have no notion of "pause requested by printer". Octoprint has this feature ("//action: paused"), but it's important it works with all hosts.
If we are able to differentiate the hosts, then not only can we have functional serial stream while paused because of a fan error, but you will also be able to resume from the Octoprint terminal with M602 (or with the resume button if the script is configured correctly). If I can differentiate the host type, then I'll also be able to add pause/resume/stop buttons on the LCD for an Octoprint USB print.

jcwren, 2020-11-16:
@leptun, I'm speaking from a position of extreme in-expertise on G-code in general, but what about a possible G-code addition that allows the host to indicate if it can handle pauses requested by the printer or not? Defaults to the current condition ('off' or 'disabled'), and Octoprint users can enable it in the pre-print configuration stanza?

@3d-gussner, the intermittency is probably due to a wiring issue. I didn't mean to indicate the firmware was necessarily throwing the fan error incorrectly, rather that the fan tach sensor or tach wire is damaged, but the fan is still spinning. I'll try to video this, but perhaps by disconnecting the tach lead during a print you could cause the condition.

leptun, 2020-11-16:
@jcwren yes. Such a gcode would solve our problem, but this needs to be done with support from Octoprint. Doing it in the connect gcode script would be the dumb way to do it.
There is also the method the 32bit fw team used where they abuse the action commands prompt feature in octoprint and the M876 gcode to detect octoprint, but that is not great at all (even though it could be easily implemented here as well).

So the solution is creating a gcode designed specifically for letting the printer know what host is used. So this should be a feature request, not a bug report.
@DRracer please update the tags accordingly.

DRracer, 2020-11-16:
Thanks for reporting this issue, marking as 3.10 and we shall think again about identification of the host software driving our machine.

3d-gussner, 2020-11-16:
> @3d-gussner, the intermittency is probably due to a wiring issue. I didn't mean to indicate the firmware was necessarily throwing the fan error incorrectly, rather that the fan tach sensor or tach [...]

I will try to test by "breaking" the tacho signal and to reproduce your error.
- Print something via Octoprint
- Pause from Octoprint
- "Break" tacho to generate Fan error
- Try to resume from Octoprint

But if you have an unreliable tacho you should consider to fix the issue or disable the fan error function, or not?

jcwren, 2020-11-16:
The failure order should be:
* Print something via Octoprint
* "Break" tacho to generate fan error
* Try to resume from Octoprint after "fixing" tacho

I can resume from the front panel but not with an M602, which I feel I should be able to do.

You are correct that I should fix the actual cause of the problem, but I could see this occurring if something jams the fan, clear the jam, then resume from Octoprint. Unfortunately, when this occurred for me, I needed to finish the print, which had about 3 hours left. This caused me to run down to the basement either 3 times in 15 minutes or once every 15 minutes, depending on often the falsing was occuring. Granted that the exercise is good for me, it would have more convenient to resume from the laptop upstairs :)

closed 2023-11-02 by github-actions[bot]