prusa3d/Prusa-Firmware #2891: ???[BUG] MK3S Filament Sensor isn´t recognized as v0.4 or newer

Markus-1984, 2020-11-15:
**Printer type** - MK3S
**Printer firmware version** -  3.9.0 - 3.9.1 - 3.9.2

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
The Filament Sensor is recognized as v0.3 or older instead of 0.4 or newer.
It is the red one with a v0.4 print on it.

I´ve received my MK3S a few week ago. First i´ve updated the firmware to 3.9.1 as it´s decribed in the manual.
Accidently I choose the Firmware for the MK3 instead of the MK3S.
The first time setup routine runs through without having any issues.
As I mentioned that I´m using the wrong firmware, because the printer says the data is sliced for another printer, I updated the right firmware. The setup routine worked fine, till it wants to calibrate the first layer and tries to load the filament.
The printer tried several times to load the filament, but don´t recognize that the filament is loaded. So it extrudes filament trough the nozzle and do a heap of filament on the bed until i reset the printer.
When i connect Pronterface, i can see that the filament sensor is set to v0.3 or older.
Unfortualy I don´t have the Settings -> HW Setup -> Fsensor Detection menue.
I´ve tried different types of firmwares like 3.9.0, 3.9.1 and 3.9.2 for the MK3S. In all of these, i don´t have the menue.
It´s not caused by the selected language, i´ve tested german and english.
After I´ve done a firmware change, I do a factory reset ALL DATA.
Then the sensor changed to unknown state. But after the inital setup routine, it´s still 0.3 or older.

With the wrong Firmware, the Printer is working well including the MMU2S. All of my prints are very well.
But i want to have the right firmware on the printer, cause it´s a MK3S and not a MK3.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Markus-1984, 2020-11-15:
I´ve tried the firmware v3.9.0. Here I got the menue Settings -> HW Setup -> Fsensor Detection. I´ve done the detection and the sensor is recognized as v0.4 or never. But unfortualy the printer failed to load the filament, too. Same as before, it produces a heap of filament on the bed when he tries to load the filament.

FlorianGerhardt, 2020-11-19:
I got the same error as you, filament sensor rev 0.4 being recognized as 0.3 or older.

Afaik, when the MMU2S is connected, there is no manual detection for the filament sensor, only the standalone MK3/S can trigger the detection. The changelog says that you have to run a self test in order to update the filament sensor values when you have a MMU2S: https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0 .

What are your voltages with and without filament? They should be "conditions: no filament: 4.2-4.6 V, inserted filament: 0.3-0.5 V"

FlorianGerhardt, 2020-11-20:
Due to issues with my MMU2S I disconnected it and the MK3S has no issues recognizing the sensor v0.4.

DanielSMT, 2021-01-02:
I get the same issue (in both FW 3.9.2 and 3.9.3 RC1) - it seems when the MMU2 is attached the new filament sensor is not detected and there is no option in HW Setup to detect it (unlike if there is no MMU2)
I recently updated my MK3S MMU2S to the new "+" version which included the new super pinda and a new filament sensor.
The filament sensor was the new red one with 0.4 on it rather than the old 0.1 that I took out.
After going through the setup wizard I still get the message that the filament sensor is 0.3 or lower.
I tried doing a full reset and updating from 3.9.2 to 3.9.3 RC1 and going through full wizard and calibration again but still doesnt detect that the sensor is version 0.4.
(The sensor seems to work the same as the original though, it would just be good if it showed the correct version, in theory I could have left the old one there when updated superpinda etc)

crix26, 2021-01-27:
I also get this issue.
I upgraded my MK3/S with MMU2S to MK3/S+. Before i had FS 0.1b and added SuperPINDA and FS 0.4.
I hat FW 3.9.3-3556 on the machine before i started the upgrade.
Filament Sensor works fine, voltage is 4.4V without and 0.4V with filament, printing also works fine.
Status says Filament Sensor Version: Unknown Status
I Loaded and Unloaded Filament
I Ran Self Check
Restarted
Status changed to "v0.3 or older"
I contacted Prusa Support, they were able to reproduce this behaviour on their machine and told me to point their developers again to this bug.

3d-gussner, 2021-01-27:
At this moment the firmware sets the IR sensor to "v0.3 or older" with a MMU2S detected, and doesn't check which hardware version it is.
So if you have the IR sensor v0.4 or newer installed it will show:
- MK3S/+ as "v0.4 or newer"
- MK3S/+ with MMU2S as "v0.3 or older"

This "minor" issue is known and may be addressed in one of the following versions.

Do you have to worry about?
No.
The MK3S/+ with MMU2S uses the MMU2S FINDA as filament runout sensor and the IR sensor is in this combination "only" to help with the loading/unloading of the filament during the changes.

A standalone MK3S/+ uses the IR sensor as filament runout sensor, and the "v0.4 or newer" added some additional  features to detect if wiring is correct or the IR sensor behaves within defined tolerances.

Also the "Unknown Status" is also a known "minor" issue and may be addressed in one of the following versions.

"Unknown Status" is shown if the MK3S/+ doesn't receive an update from the MMU2S FINDA status in a defined time.
The problem is find the correct behaviour for all operations:
- The timeout has to be short enough to detect filament runout, so it can be pulled out correctly
- On the other hand there are operations that may take more time to complete

Hope I could help to understand the issue a bit and that you have to worry less about it.

clfaye, 2021-03-03:
I have this same issue (new v0.4 filament sensor not recognized).  When you combine this issue with the fact that the filament sensor is extremely unreliable in the MMU2S already - it's very frustrating.  The filament sensor is one of the worst parts of this printer that I otherwise really like.  The MMU2S is extremely unreliable and the filament sensor is one of the main reasons.

Markus-1984, 2021-03-28:
Today i´ve updated the Firmware of my Mk3s to 3.9.3 and the IR Sensor is recongnised as 0.4 or newer.
But I still have the same load failure with the MMU2s.
When the printer tries to load the filament, the MMU push the filament but don´t stop when it reaches the nozzle. Filamentsensor is working, but the MMU don´t stop to push until i reset the printer.
Please fix this issue, cause i´m very unhappy with this.
I´ll try to modify the 3.9.3 for the MK3s like i´ve done it before and hope that it works. In the past, i´ve changed manualy the IR Sensor to the "old" version and it works.

closed 2022-01-23 by Markus-1984